pengutronix/genimage #2: Multi-component directory names in genimage.cfg fail

bbarnett, 2016-03-14:
I'm using a genimage.cfg file that looks like this:
===========
image ad_emmc.img {
        hdimage {}
        partition rootfsA {
                in-partition-table = "yes"
                partition-type = 0x83
                image = "rootfs.ext4"
                bootable = "yes"
                size = 1000M
        }
        partition rootfsB {
                in-partition-table = "yes"
                partition-type = 0x83
                image = "rootfs.ext4"
                bootable = "yes"
                size = 1000M
        }
        partition services {
                in-partition-table = "yes"
                partition-type = 0x83
                image = "services.ext4"
                size = 2000M
        }
        partition var {
                in-partition-table = "yes"
                partition-type = 0x83
                image = "var.ext4"
                size = 500M
        }
        partition skills {
                in-partition-table = "yes"
                partition-type = 0x83
                image = "skills.ext4"
                size = 10000M
        }
}

image rootfs.ext4 {
        name = "rootfs"
        ext4 {
                label = "rootfs"
        }
        size = 1000M
        mountpoint = "/"
}

image var.ext4 {
        name = "var"
        ext4 {
                label = "var"
        }
        size = 500M
        mountpoint = "/var"
}

image services.ext4 {
        name = "services"
        ext4 {
                label = "services"
        }
        size = 2000M
        mountpoint = "/usr/local"
}

image skills.ext4 {
        name = "skills"
        ext4 {
                label = "skills"
        }
        size = 100M
        mountpoint = "/opt"
}
===========
And the post-image.sh looks like this:
===========
#!/bin/sh

GENIMAGE_CFG="board/nvidia/avionic/genimage.cfg"
GENIMAGE_ROOT="${BUILD_DIR}/genimage.root"
GENIMAGE_TMP="${BUILD_DIR}/genimage.tmp"

# Cleanup leftovers
rm -rf ${GENIMAGE_ROOT}
rm -rf ${GENIMAGE_TMP}

# Provide rootfs
mkdir -p ${GENIMAGE_ROOT}
fakeroot tar -C ${GENIMAGE_ROOT} -xf ${BINARIES_DIR}/rootfs.tar

fakeroot genimage \
        --rootpath "${GENIMAGE_ROOT}" \
	--tmppath "${GENIMAGE_TMP}" \
	--inputpath "${BASE_DIR}" \
	--outputpath "${BINARIES_DIR}" \
	--loglevel=10 \
	--config "${GENIMAGE_CFG}"

RET=${?}

IMAGES="${1}"
exit ${RET}
===============

When post-image.sh runs, I get the following output:
===============
cmd: mkdir -p /home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp
cmd: rm -rf /home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp/*
cmd: mkdir -p /home/blair/work/testing/buildroot.jibo/output/images
cmd: mkdir -p /home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp
cmd: cp -a /home/blair/work/testing/buildroot.jibo/output/build/genimage.root /home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp/root
cmd: mv /home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp/root/var /home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp
cmd: mkdir /home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp/root/var
cmd: chmod --reference=/home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp/var /home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp/root/var
cmd: chown --reference=/home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp/var /home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp/root/var
**cmd: mv /home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp/root/usr/local /home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp**
cmd: mkdir /home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp/root/usr/local
cmd: chmod --reference=/home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp/usr/local /home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp/root/usr/local
chmod: failed to get attributes of ‘/home/blair/work/testing/buildroot.jibo/output/build/genimage.tmp/usr/local’: No such file or directory
===============

So the  mv command does what it's supposed to do, but the script was expecting it to move the directory tree when it just move the last directory.

When faced with "/usr/local" the 'mv" command will just recreate the "local" directory.

bbarnett, 2016-03-14:
Wow, that formatting is pretty weird, but I'm not going to do anything about it.

michaelolbrich, 2017-12-04:
Should be fixed by #21.

closed 2017-12-04 by michaelolbrich