pengutronix/genimage #3: create folder on fat partition?

mllapps, 2016-03-24:
I want to create a image for my sdcard/usb-stick. Some files on the first boot partition should be at subfolder. Is it possible to create a vfat partition and add the files to a subfolder instead of the root path?

mllapps, 2016-04-20:
Is it possible to write the files to a subfolder?

oscargomezf, 2016-04-27:
Hi, I've got the same question. I'm working with the raspberry pi3 and I need to create a folder on the boot.vfat partition called overlays and include in this folder a .dtb file.

Do you have any update? or Did you find any solution?

Best regards.

mllapps, 2016-04-27:
Hy,

yes I found a solution but currently not with genimage. I use the same steps like genimage works. The steps are:

1. create a image (using `dd`)
2. formate the image (using `mkfs.vfat`)
3. create a folder for example "overlay" (using `mkdir`) 
4. copy the folder to the image (using `mcopy`)
5. copy all files you want to the folder on the image (using `mcopy`)

I think genimage currently do not support subfolders because you have to copy a folder before you are able to add files to it.

mllapps, 2016-05-24:
I found the solution with genimage.

clems71, 2016-05-25:
Hi all!
By reading the source code, I tried this:

    image boot.vfat {
      vfat {
        files = {
          "bcm2708-rpi-b-plus.dtb",
          "rpi-firmware/bootcode.bin",
          "rpi-firmware/cmdline.txt",
          "rpi-firmware/config.txt",
          "rpi-firmware/fixup.dat",
          "rpi-firmware/start.elf",
          "kernel-marked/zImage"
        }
    
        file overlays/dwc2.dtbo {
          image = "rpi-firmware/overlays/dwc2.dtbo"
        }
    
      }
      size = 32M
    }

I tested it and it works properly! The file dwc2.dtbo ends in a directory overlays on VFAT partition.

h6w, 2017-02-01:
This only seems to work for vfat.  Is there an equivalent in ext2/3/4?

sydis, 2017-04-25:
@h6w I've given an answer in issue #5

michaelolbrich, 2017-11-07:
There are two ways to do this. As @clems71 noted above, you can specify the full path for each file with individual file sections.
Or you can prepare the content in one directory and use `--rootpath`. This should work just like it does for ext2/3/4 as described in issue #5. Note: this only works is no `file` or `files` entry is specified.

closed 2017-11-07 by michaelolbrich