pengutronix/genimage #4: Syntax error on configure

h6w, 2017-01-12:
When I try to compile genimage on my Ubuntu 16.04 LTS with libconfuse-dev installed, I get this:

```
$ ./autogen.sh
configure.ac:78: warning: AC_LANG_CONFTEST: no AC_LANG_SOURCE call detected in body
../../lib/autoconf/lang.m4:194: AC_LANG_CONFTEST is expanded from...
../../lib/autoconf/general.m4:2662: _AC_LINK_IFELSE is expanded from...
../../lib/autoconf/general.m4:2679: AC_LINK_IFELSE is expanded from...
../../lib/m4sugar/m4sh.m4:606: AS_IF is expanded from...
../../lib/autoconf/general.m4:2032: AC_CACHE_VAL is expanded from...
../../lib/autoconf/general.m4:2053: AC_CACHE_CHECK is expanded from...
m4/attributes.m4:87: CC_CHECK_LDFLAGS is expanded from...
configure.ac:78: the top level
configure.ac:79: warning: AC_LANG_CONFTEST: no AC_LANG_SOURCE call detected in body
../../lib/autoconf/lang.m4:194: AC_LANG_CONFTEST is expanded from...
../../lib/autoconf/general.m4:2662: _AC_LINK_IFELSE is expanded from.
........
configure.ac:79: the top level
configure.ac:9: installing `build-aux/install-sh'
configure.ac:9: installing `build-aux/missing'
Makefile.am: installing `build-aux/depcomp'

----------------------------------------------------------------
Initialized build system. For a common configuration please run:
----------------------------------------------------------------

./configure CFLAGS='-g -O0' --prefix=/usr

root@a2be79f92dca:/buildpid/genimage/genimage-master# ./configure CFLAGS='-g -O0' --prefix=/usr
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
.......
checking whether to enable debugging... no
checking whether to hide internal symbols... yes
checking if gcc -std=gnu99 supports -Wl,--as-needed flag... yes
checking if gcc -std=gnu99 supports -Wl,--gc-sections flag... yes
./configure: line 5531: syntax error near unexpected token `CONFUSE,'
./configure: line 5531: `PKG_CHECK_MODULES(CONFUSE, libconfuse)'
$ 
```
This is with autoconf 2.68

Is there something I'm missing?

michaelolbrich, 2017-11-07:
`PKG_CHECK_MODULES` should have been expanded. Maybe `/usr/share/aclocal/pkg.m4` does not exist on your system. Is `pkg-config` installed? If not, then install it and rerun `autogen.sh`.

h6w, 2017-11-12:
Yes, pkg-config is installed and /usr/share/aclocal/pkg.m4 exists, but there is no confuse.pc file in /usr/share/pkgconfig and PKG_CHECK_MODULES is empty.

michaelolbrich, 2017-11-13:
Hmm, can trigger this problem when autoreconf fails to find pkg.m4. Where does your autoconf/automake come from? Is it manually compiled or the system version?

michaelolbrich, 2018-12-13:
No response for a long time. Please reopen with more information if the problem still exists.

closed 2018-12-13 by michaelolbrich