pengutronix/genimage #5: Files and folders in ext2 partition

silart-teplomonitor, 2017-04-19:
In vfat partition you can create folders and files in the following way:
image boot.vfat {
  vfat {
    files = {
      "bcm2708-rpi-b-plus.dtb",
      "rpi-firmware/bootcode.bin",
      "rpi-firmware/cmdline.txt",
      "rpi-firmware/config.txt",
      "rpi-firmware/fixup.dat",
      "rpi-firmware/start.elf",
      "kernel-marked/zImage"
    }

    file overlays/dwc2.dtbo {
      image = "rpi-firmware/overlays/dwc2.dtbo"
    }

  }
  size = 32M
}

I want to create files and folders in ext2 partitions. How can I do it? This way is not working:
image home.ext2 {
	name = "home"
	ext2 {
    files = {
      "bcm2708-rpi-b-plus.dtb",
      "rpi-firmware/bootcode.bin"
    }
}
	size = 128M
	mountpoint = "/home"
}

sydis, 2017-04-19:
Hi,
Files need to be located somewhere on your disk, say in
`/somewhere/home/`
The call to genimage needs a parameter
`--rootpath "/somewhere"`
and another one with your script
`--config "example_config.cfg"`
The example_config.cfg file contains :
```
image home_image.ext2 {
  ext2 {
  }
  size = 128M
  mountpoint = "/home"
}
```

Hope this would work in your case and be helpful !

michaelolbrich, 2017-11-07:
As @sydis showed you can use `--rootpath` and `mountpoint = ` to add a directory tree to the filesystem. This works for all filesystem image types.
Only `vfat` has the extra `file`/`files` options. It was implemented because in many cases vfat partition contain only a small number of files (e.g. the rpi boot partition).

closed 2017-11-07 by michaelolbrich