pengutronix/genimage #6: hdimage(disk.img): fwrite 4096: Invalid argument

CatsMum, 2017-04-25:
I'm getting the above error message using genimage with the config file shown below. I'm baffled. Any ideas?

image disk.img {

  hdimage {
  }

  partition boot {
    in-partition-table = "no"
    image = "boot.img"
    offset = 0
    size = 512
    size = 1048576
  }

  partition grub {
    image = "grub.img"
    offset = 1048576
    size = 33554432
    bootable = "yes"
  }

}

Below is the directory with the source files in it - as you can see I'm using it with buildroot.

drwxr-xr-x 2 nicole nicole 4.0K Apr 25 20:25 ./
drwxr-xr-x 6 nicole nicole 4.0K Apr 25 17:24 ../
lrwxrwxrwx 1 nicole nicole   88 Apr 25 20:24 boot.img -> /home/nicole/projects/router-builder/buildroot/output/host/usr/lib/grub/i386-pc/boot.img
-rw-r--r-- 1 nicole nicole 4.6M Apr 25 17:59 bzImage
-rw-r--r-- 1 nicole nicole 1.2M Apr 25 20:25 disk.img
-rw-r--r-- 1 nicole nicole 114K Apr 25 18:01 grub-eltorito.img
-rw-r--r-- 1 nicole nicole 114K Apr 25 18:01 grub.img
-rw-r--r-- 1 nicole nicole  16G Apr 25 20:25 rootfs.ext2
lrwxrwxrwx 1 nicole nicole   11 Apr 25 20:25 rootfs.ext4 -> rootfs.ext2

sydis, 2017-04-25:
It seems you have written the "size" too much times in partition boot. Once is enough !
If your boot.img is only 512 bytes, that would be the right one to keep. To stay on a 1M alignment, write offset = 1M in partition grub. 1M is precisely 1024 x 1024.

CatsMum, 2017-04-25:
Isn't it funny how you don't see obvious things until someone points it out? Thank you for your patience and polite reply. Thanks for the pointer - I've been reading the code and noticed that the program understood these multiplier suffices. I'll stick with the numbers, though, because they're calculated (with some difficulty) in a bash script and applied to the config file. It was the template for this process that was in error.

closed 2017-04-25 by CatsMum