pengutronix/genimage #7: hdimage(disk.img): fwrite 4096: Invalid argument - previous issue reprised

CatsMum, 2017-04-25:
I corrected my config file so that he boot area is 1M in size (shown below), but am still getting this error. Any ideas, anyone?

```
partition boot {
    in-partition-table = "no"
    image = "boot.img"
    offset = 0
    size = 1048576
}
```

sydis, 2017-04-26:
Hi again
I remember in your file list that boot.img is actually a link.
What happens if you make a copy in your image folder instead of a link ?

That's just for advice, but I seriously doubt that your boot.img is more than 512 B. And does your grub really need to be in the partition table ? I've done this before with buildroot :
```
image drive.img {
  hdimage {
  }
  size = 4G
  partition boot {
    in-partition-table = "no"
    image = "boot.img"
    offset = 0
    size = 512
  }
  partition grub {
    in-partition-table = "no"
    image = "grub.img"
    offset = 512
    size = 1048064
  }
  partition other {
    image = "other.img"
    offset = 1M
    size = 32M
  }
<and so on>
}
```

CatsMum, 2017-04-26:
Thanks for all your help and patience. I resolved this by rebooting (gosh!) my PC. Somehow it had got itself into a state where this particular system call in this particular program was producing this anomalous status.

About 35 years ago I read an article in Computing (a weekly trade paper in the UK) about the value of programming in pairs. You had to explain your thinking, in detail, to your partner, and in doing so revealed the flaws in your thinking. The author finished by suggesting that in the absence of a partner, a life-sized cardboard cut-out would do.

closed 2017-04-26 by CatsMum