pengutronix/genimage #8: Fakeroot and SUID

behnamgolds, 2017-06-22:
Hi,
When I execute  genimage with fakeroot the suid bits are not preserved ,but with sudo it works fine. Am I missing something here? Cause I thought fakeroot was the safest way to go. And needless to say. the root filesystem is extracted  from a tar.bz2 archive (with -p and --same-owner tar options)  under fakeroot . I also checked the files after exiting the fakeroot, the suid bits are all there (although with my normal user as the owner).

With fakeroot:
```
ls -l /bin/su
-rwxr-xr-x 1 root root 43924 May 28 14:47 /bin/su
```

With sudo:
```
ls -l /bin/su
-rwsr-xr-x 1 root root 43924 May 28 15:12 /bin/su
```

On the other hand ,when using the [Yocto project](https://www.yoctoproject.org/)  as a normal user we get images containing files with suid bits set. So that's possible to set suid as a normal user under a fakeroot(ish?) environment.

michaelolbrich, 2017-11-07:
Can you provide more details (config file, exact commands / scripts)? I use fakeroot and genimage all the time and I've never seen this problem.

gazoo74, 2017-11-24:
I think you lost the SUID bit because you run many fakeroot commands and you lost the environment at each instance.

You can load/save the environment using `-i env.txt`/`-s env.txt`.

michaelolbrich, 2018-12-13:
No response for a long time. Please reopen with more information if the problem still exists.

closed 2018-12-13 by michaelolbrich