pengutronix/genimage #11: "size = 0" gives error "hdimage(sdcard.img) part data size must not be zero"

Sjoerd82, 2017-10-13:
I'm trying to create an empty FAT partition in the remaining space on an SD card.

My config file:
```
image boot.vfat {
  vfat {
    files = {
      "bcm2708-rpi-b-plus.dtb",
      "bcm2708-rpi-0-w.dtb",
      "rpi-firmware/bootcode.bin",
      "rpi-firmware/cmdline.txt",
      "rpi-firmware/config.txt",
      "rpi-firmware/fixup.dat",
      "rpi-firmware/start.elf",
      "zImage"
    }

    file overlays/hifiberry-dac.dtbo {
      image = "rpi-firmware/overlays/hifiberry-dac.dtbo"
    }

  }
  size = 32M
}

image sdcard.img {
  hdimage {
  }

  partition boot {
    partition-type = 0xC
    bootable = "true"
    image = "boot.vfat"
  }

  partition rootfs {
    partition-type = 0x83
    image = "rootfs.ext4"
  }

  partition data {
    partition-type = 0xC
    size = 0
  }

}
```

I cannot get it to accept 0 in order to use up the remaining space, giving me the error:
`hdimage(sdcard.img) part data size must not be zero`

I am creating this using Buildroot on an 8GB SD card, with the rootfs set to 1GB in the buildroot config.

kyle7119, 2017-11-02:
You need to specify a size for the sdcard.img. Genimage cannot determine how much remaining space to fill if it doesn't know the final size of the image.

```
image boot.vfat {
  vfat {
    files = {
      "bcm2708-rpi-b-plus.dtb",
      "bcm2708-rpi-0-w.dtb",
      "rpi-firmware/bootcode.bin",
      "rpi-firmware/cmdline.txt",
      "rpi-firmware/config.txt",
      "rpi-firmware/fixup.dat",
      "rpi-firmware/start.elf",
      "zImage"
    }

    file overlays/hifiberry-dac.dtbo {
      image = "rpi-firmware/overlays/hifiberry-dac.dtbo"
    }

  }
  size = 32M
}

image sdcard.img {
  hdimage {
  }
  size = 2G
  partition boot {
    partition-type = 0xC
    bootable = "true"
    image = "boot.vfat"
  }

  partition rootfs {
    partition-type = 0x83
    image = "rootfs.ext4"
  }

  partition data {
    partition-type = 0xC
    size = 0
  }

}
```

gazoo74, 2017-12-11:
Hum... I don't think it works either.

I made a patch for that.

When the size of the last partition is unset (or set to 0) *and* the size of the hdimage is set, the hdimage uses the remaining space as the size of the last partition.

I have tested it only in the case of an extended partition.

@michaelolbrich: what do you think about this feature?

michaelolbrich, 2017-12-12:
1. I'd like something more explicit, maybe a `fill` option or something like that.
2. Use `pad_file()` to do this. Like this, I think (untested):
```
ret = pad_file(image, NULL, imageoutfile(image), image->size,                                                                             
        0, MODE_OVERWRITE);                                                                                                               
```

yannickbt64, 2019-03-06:
Is there a way to make it work.
Right now if I specify a size for hdimage (I figured I would need to since its just common sense), "fill" mode doesn't work with "size = 0". Frankly its annoying especially since it says in the doc size = 0 should have this expected effect.
Having to calculate the sum of the partition sizes each time I edit the config plain sucks.

michaelolbrich, 2019-05-04:
@yannickbt64 the documentation is incorrect :-/. Take a look at #59. I think that should implement what you need.

michaelolbrich, 2019-05-11:
Fix is now in master.

closed 2019-05-11 by michaelolbrich