pengutronix/genimage #31: Add all subfolder's files to the root of the image

substring, 2018-03-22:
Hello,

I've been trying to add everything from a folder to the root of a vfat image without success. How can I achieve this ? I'm thinking about something as simple as `files { "folder/" }` or `files { "folder/*" }` The first syntaxe creates `folder` in the image, the second one gives an error. This does work for subfolders (`files { "folder/subfolder" }` is added as `subfolder`). 

Please note that syntax `files { "folder/subfolder" }` produces output `adding file 'folder/subfolder' as 'folder/subfolder' ...` whereas it means `adding file 'folder/subfolder' as 'subfolder' ...`

Using v9 genimage from Buildroot.

michaelolbrich, 2018-03-23:
If you don't use file / files then the usual rootpath / mountpoint. I've created #32 to clarify this in the README (among other things).

substring, 2018-03-24:
I'm not sure I undserstand. Let's take a practical example with [this .cfg](https://github.com/buildroot/buildroot/blob/master/board/raspberrypi/genimage-raspberrypi3.cfg) from Buildroot. What would the syntax be to include everything from the `rpi-firmware` [folder](https://github.com/buildroot/buildroot/blob/master/board/raspberrypi/genimage-raspberrypi3.cfg#L6-L11) to the root of the image file instead of specifying files and folders one by one ?

michaelolbrich, 2018-11-16:
Don't add the `files` to the vfat image and set `rootpath` to the rpi-firmware folder.

closed 2019-05-11 by michaelolbrich