pengutronix/genimage #40: GPT support for genimage

lp35, 2018-08-28:
Hi,

UEFI platform are more and more used even in embedded world. Currently using buildroot as my distro, it is actually not possible to generate a GPT compatible image.

I think it would be really useful to the community, any hope for implementing this?

http://buildroot-busybox.2317881.n4.nabble.com/genimage-with-GPT-partition-td202341.html

Have a nice day, and thank you for maintaining this project.
BR

lp35, 2018-10-10:
Hi, I am bumping this. Thanks,

michaelolbrich, 2018-12-13:
I would like to see support for GPT as well, but it is not very high on my TODO list right now. The problem with GPT is that it has some data at the end of the device (At least that's my current understanding). This means that:

1. The config must contain the exact device size. Especially with SD-Cards this can be a Problem. You cannot just use another one.
2. This means always overwriting the whole device. That's often an order of magnitude more than the actual content. Or we need some kind of sparse format and tool to write it.

All that makes it not very useful for my use-cases, so my priority is low.

If someone wants to implement this, then I'll gladly review any patches. GPT is rather complex, so I think we should use some external implementation for this. Maybe `libfdisk`?

PR #51 (2019-02-15): hdimage: Add GPT support
Fixes #40.
closed 2019-02-15 by michaelolbrich