pengutronix/genimage #49: after creating an Image the file systems are corrupt

EugenWiens, 2018-12-21:
genimage.config
`
image PRODUCT-emmc-dev.img {
	hdimage {
		disk-signature = 0x741ac374
	}

	partition firmware-slot-0 {
		image = "PRODUCT-rootfs-dev-board.ext4"
		partition-type = 0x83
		size = 6700M
	}

	partition firmware-slot-1 {
		image = "PRODUCT-rootfs-dev-board.ext4"
		partition-type = 0x83
		size = 6700M
	}
}`

After the file **PRODUCT-emmc-dev.img** is written to the eMMC the system doesn't start. See attached  Log. 
`
[    3.913317] EXT4-fs: Warning: mounting with data=journal disables delayed allocation and O_DIRECT support!
[    3.929445] JBD2: no valid journal superblock found
[    3.934340] EXT4-fs (mmcblk1p1): error loading journal
[    3.939951] List of all partitions:
[    3.943476] 1f00             128 mtdblock0 
[    3.943481]  (driver?)
[    3.950042] 1f01             128 mtdblock1 
[    3.950049]  (driver?)
[    3.956606] 1f02             128 mtdblock2 
[    3.956611]  (driver?)
[    3.963166] 1f03             128 mtdblock3 
[    3.963171]  (driver?)
[    3.969728] b300        15269888 mmcblk1 
[    3.969733]  driver: mmcblk
[    3.976563]   b301         6860800 mmcblk1p1 741ac374-01
[    3.976567] 
[    3.983381]   b302         6860800 mmcblk1p2 741ac374-02
[    3.983385] 
[    3.990208] b318            4096 mmcblk1rpmb 
[    3.990212]  (driver?)
[    3.996946] b310            4096 mmcblk1boot1 
[    3.996952]  (driver?)
[    4.003767] b308            4096 mmcblk1boot0 
[    4.003771]  (driver?)
[    4.010584] No filesystem could mount root, tried: 
[    4.010589]  ext4
[    4.015477] 
[    4.018906] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,1)
[    4.018922] CPU2: stopping
[    4.018934] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 4.14.69-20181017-1-rt43 #3
[    4.018937] Hardware name: Freescale i.MX6 Quad/DualLite (Device Tree)
[    4.018940] Backtrace: 
[    4.018961] [<c010e354>] (dump_backtrace) from [<c010e684>] (show_stack+0x20/0x24)
[    4.018970]  r7:ffffe000 r6:60000193 r5:00000000 r4:c10ba070
[    4.018986] [<c010e664>] (show_stack) from [<c0a7b984>] (dump_stack+0x90/0xa4)
[    4.019000] [<c0a7b8f4>] (dump_stack) from [<c01125f0>] (handle_IPI+0x400/0x45c)
[    4.019008]  r7:ffffe000 r6:00000000 r5:00000004 r4:c10dd2d0
[    4.019022] [<c01121f0>] (handle_IPI) from [<c01015dc>] (gic_handle_irq+0x9c/0xa0)
[    4.019031]  r10:00000001 r9:f4001100 r8:ef161ee8 r7:f4000100 r6:f400010c r5:c1066980
[    4.019035]  r4:c1006b2c
[    4.019043] [<c0101540>] (gic_handle_irq) from [<c010f34c>] (__irq_svc+0x6c/0xbc)
[    4.019051] Exception stack(0xef161ee8 to 0xef161f30)
[    4.019059] 1ee0:                   00000000 ef7c0540 00000001 ef160000 eeaff014 00000000
[    4.019068] 1f00: ef8bc037 00000000 00000001 c10c151c 00000001 ef161f74 ef161ea8 ef161f38
[    4.019074] 1f20: c01a5c90 c07b48bc 20000113 ffffffff
[    4.019081]  r9:ef160000 r8:00000001 r7:ef161f1c r6:ffffffff r5:20000113 r4:c07b48bc
[    4.019098] [<c07b4740>] (cpuidle_enter_state) from [<c07b4cb0>] (cpuidle_enter+0x24/0x28)
[    4.019106]  r10:c0d4a870 r9:c10c09d8 r8:c0fc4810 r7:ef7bc818 r6:c10064fc r5:c10064b0
[    4.019109]  r4:ffffe000
[    4.019120] [<c07b4c8c>] (cpuidle_enter) from [<c016d57c>] (call_cpuidle+0x40/0x58)
[    4.019129] [<c016d53c>] (call_cpuidle) from [<c016d828>] (do_idle+0x168/0x200)
[    4.019140] [<c016d6c0>] (do_idle) from [<c016dbd4>] (cpu_startup_entry+0x28/0x2c)
[    4.019148]  r10:00000000 r9:412fc09a r8:1000406a r7:c10dd2e0 r6:10c0387d r5:00000002
[    4.019151]  r4:00000084
[    4.019164] [<c016dbac>] (cpu_startup_entry) from [<c0111f5c>] (secondary_start_kernel+0x184/0x1a8)
[    4.019174] [<c0111dd8>] (secondary_start_kernel) from [<1010166c>] (0x1010166c)
[    4.019179]  r5:00000051 r4:3f14806a
[    4.019184] CPU3: stopping
[    4.019191] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 4.14.69-20181017-1-rt43 #3
[    4.019194] Hardware name: Freescale i.MX6 Quad/DualLite (Device Tree)
[    4.019196] Backtrace: 
[    4.019209] [<c010e354>] (dump_backtrace) from [<c010e684>] (show_stack+0x20/0x24)
[    4.019217]  r7:ffffe000 r6:60000193 r5:00000000 r4:c10ba070
[    4.019225] [<c010e664>] (show_stack) from [<c0a7b984>] (dump_stack+0x90/0xa4)
[    4.019235] [<c0a7b8f4>] (dump_stack) from [<c01125f0>] (handle_IPI+0x400/0x45c)
[    4.019241]  r7:ffffe000 r6:00000000 r5:00000004 r4:c10dd2d0
[    4.019250] [<c01121f0>] (handle_IPI) from [<c01015dc>] (gic_handle_irq+0x9c/0xa0)
[    4.019258]  r10:00000001 r9:f4001100 r8:ef163ee8 r7:f4000100 r6:f400010c r5:c1066980
[    4.019261]  r4:c1006b2c
[    4.019268] [<c0101540>] (gic_handle_irq) from [<c010f34c>] (__irq_svc+0x6c/0xbc)
[    4.019272] Exception stack(0xef163ee8 to 0xef163f30)
[    4.019281] 3ee0:                   00000000 ef7d1540 00000001 ef162000 eeafdf27 00000000
[    4.019290] 3f00: ef8bc037 00000000 00000001 c10c151c 00000001 ef163f74 ef163ea8 ef163f38
[    4.019295] 3f20: c01a5c90 c07b48bc 20000013 ffffffff
[    4.019304]  r9:ef162000 r8:00000001 r7:ef163f1c r6:ffffffff r5:20000013 r4:c07b48bc
[    4.019316] [<c07b4740>] (cpuidle_enter_state) from [<c07b4cb0>] (cpuidle_enter+0x24/0x28)
[    4.019324]  r10:c0d4a870 r9:c10c09d8 r8:c0fc4810 r7:ef7cd818 r6:c10064fc r5:c10064b0
[    4.019326]  r4:ffffe000
[    4.019336] [<c07b4c8c>] (cpuidle_enter) from [<c016d57c>] (call_cpuidle+0x40/0x58)
[    4.019344] [<c016d53c>] (call_cpuidle) from [<c016d828>] (do_idle+0x168/0x200)
[    4.019351] [<c016d6c0>] (do_idle) from [<c016dbd4>] (cpu_startup_entry+0x28/0x2c)
[    4.019359]  r10:00000000 r9:412fc09a r8:1000406a r7:c10dd2e0 r6:10c0387d r5:00000003
[    4.019362]  r4:00000084
[    4.019371] [<c016dbac>] (cpu_startup_entry) from [<c0111f5c>] (secondary_start_kernel+0x184/0x1a8)
[    4.019379] [<c0111dd8>] (secondary_start_kernel) from [<1010166c>] (0x1010166c)
[    4.019383]  r5:00000051 r4:3f14806a
[    4.019388] CPU1: stopping
[    4.019398] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.14.69-20181017-1-rt43 #3
[    4.019402] Hardware name: Freescale i.MX6 Quad/DualLite (Device Tree)
[    4.019405] Backtrace: 
[    4.019419] [<c010e354>] (dump_backtrace) from [<c010e684>] (show_stack+0x20/0x24)
[    4.019428]  r7:ffffe000 r6:60000193 r5:00000000 r4:c10ba070
[    4.019437] [<c010e664>] (show_stack) from [<c0a7b984>] (dump_stack+0x90/0xa4)
[    4.019448] [<c0a7b8f4>] (dump_stack) from [<c01125f0>] (handle_IPI+0x400/0x45c)
[    4.019456]  r7:ffffe000 r6:00000000 r5:00000004 r4:c10dd2d0
[    4.019464] [<c01121f0>] (handle_IPI) from [<c01015dc>] (gic_handle_irq+0x9c/0xa0)
[    4.019472]  r10:00000001 r9:f4001100 r8:ef15fee8 r7:f4000100 r6:f400010c r5:c1066980
[    4.019475]  r4:c1006b2c
[    4.019483] [<c0101540>] (gic_handle_irq) from [<c010f34c>] (__irq_svc+0x6c/0xbc)
[    4.019487] Exception stack(0xef15fee8 to 0xef15ff30)
[    4.019495] fee0:                   00000000 ef7af540 00000001 ef15e000 ef4898cf 00000000
[    4.019503] ff00: ef8bc184 00000000 00000001 c10c151c 00000001 ef15ff74 ef15fea8 ef15ff38
[    4.019508] ff20: c01a5c90 c07b48bc 20000013 ffffffff
[    4.019516]  r9:ef15e000 r8:00000001 r7:ef15ff1c r6:ffffffff r5:20000013 r4:c07b48bc
[    4.019528] [<c07b4740>] (cpuidle_enter_state) from [<c07b4cb0>] (cpuidle_enter+0x24/0x28)
[    4.019536]  r10:c0d4a870 r9:c10c09d8 r8:c0fc4810 r7:ef7ab818 r6:c10064fc r5:c10064b0
[    4.019539]  r4:ffffe000
[    4.019548] [<c07b4c8c>] (cpuidle_enter) from [<c016d57c>] (call_cpuidle+0x40/0x58)
[    4.019556] [<c016d53c>] (call_cpuidle) from [<c016d828>] (do_idle+0x168/0x200)
[    4.019563] [<c016d6c0>] (do_idle) from [<c016dbd4>] (cpu_startup_entry+0x28/0x2c)
[    4.019571]  r10:00000000 r9:412fc09a r8:1000406a r7:c10dd2e0 r6:10c0387d r5:00000001
[    4.019574]  r4:00000084
[    4.019583] [<c016dbac>] (cpu_startup_entry) from [<c0111f5c>] (secondary_start_kernel+0x184/0x1a8)
[    4.019592] [<c0111dd8>] (secondary_start_kernel) from [<1010166c>] (0x1010166c)
[    4.019596]  r5:00000051 r4:3f14806a
`

The whole workflow is working with the partition size of 1808M. Also if I write the file **PRODUCT-rootfs-dev-board.ext4** separately to the partitions the system start correctly. 

What am I doing wrong ? Thank you for any suggestion.

michaelolbrich, 2019-05-11:
Hmm, I have no idea what might cause this.
Can you try the current master?
And does the existence of of the second partition make a difference?
What about adding a alignment (e.g. `align = 1M`) to the hdimage section?

EugenWiens, 2019-05-14:
I will testing this points, but I'm currently busy. I will come back later to this Problem.

closed 2024-11-29 by michaelolbrich