pengutronix/genimage #67: Attempting to specify /root as a mountpoint fails

thirtythreeforty, 2019-08-03:
I have tried to add a separate partition mounted at `/root`:
```
image roothome.ubifs {
	ubifs {
		max-size = 2M
	}
	flashtype = w25q64
	mountpoint = /root
}
```
This produces an error:
```
INFO: cmd: "mkdir -p "/.../buildroot/output/build/genimage.tmp"" (stderr):
INFO: cmd: "rm -rf "/.../buildroot/output/build/genimage.tmp"/*" (stderr):
INFO: cmd: "mkdir -p "/.../buildroot/output/images"" (stderr):
INFO: cmd: "mkdir -p "/.../buildroot/output/build/genimage.tmp"" (stderr):
INFO: cmd: "cp -a "/.../buildroot/output/target" "/.../buildroot/output/build/genimage.tmp/root"" (stderr):
INFO: cmd: "mv "/.../buildroot/output/build/genimage.tmp/root/home" "/.../buildroot/output/build/genimage.tmp/home"" (stderr):
mv: cannot stat '/.../buildroot/output/build/genimage.tmp/root/home': No such file or directory
make: *** [Makefile:815: target-post-image] Error 1
```

It appears genimage is copying the various partitions into the same folder as the root filesystem, which is contained in a folder named "root." This conflicts with the mountpoint name.

I think to fix this (and not just kick the can to some other more obscure name) is to add another directory alongside:
```
genimage.tmp/
    root/ <-- the existing root directory
    mountpoints/ <-- new
        root/ <-- the overlay
```

michaelolbrich, 2019-08-05:
I think I'd prefer something like this:
```
genimage.tmp/
    root/ <-- the existing root directory
    mp-root/ <-- the overlay
```
Or any other prefix. The directory name is already generated anyways to flatten subdirectories, so this should be a very simple change in `add_mountpoint()`.

thirtythreeforty, 2019-08-05:
Yeah it wouldn't be too bad. I think an `mp-` prefix would work too... What would happen if I decided to add a mount point of `/home/myuser`? How would that look?

michaelolbrich, 2019-08-05:
This is handled by `path_sanitized()`, so (with the proposed change) this will be `genimage.tmp/mp-home-myuser/`.
In theory, there are still collisions possible (`/home/myuser/` and `/home-myuser/`) but I don't think this is really a problem.

thirtythreeforty, 2019-08-05:
Gotcha. Yes, collisions will be rare. And that "problem" (however serious you deem it) is orthogonal to this issue anyway.

So `mp-` sounds good. Would you like a PR, or would you prefer to make the change yourself?

michaelolbrich, 2019-08-05:
A PR would be great.

PR #69 (2019-09-04): Prefix mountpoints with mp- to avoid conflict with "root"
This turned out to be easier than I expected. Bodes well for me hacking on the code in the future!

Closes #67.  Thanks for this extremely useful tool.
commit f2b104aa (2019-09-04): Prefix mountpoints with mp- to avoid conflict with "root"

Specifying an image with mountpoint "/root" would cause the mountpoint's
location to be "root", which conflicts with the location for the actual
filesystem root.

Fix this by appending a prefix.

Closes #67 on GitHub.
closed 2019-09-04 by michaelolbrich