pengutronix/genimage #77: rauc files {} directive does not exclude files from rootpath

thirtythreeforty, 2019-10-09:
When using a `vfat` image, I can add a `files { ... }` directive that overrides the `mountpoint` setting and only includes the specified files:

```
image boot.vfat {
  vfat {
    files = {
      zImage
    }
  }
}
```

But when I try the same thing with a `rauc` image, it packages does not exclude `rootpath` files:

```
image update.raucb {
  rauc {
    files = {
      rootfs.squashfs,
      boot.vfat,
    }
    manifest = "(snip)"
    key = "(snip)"
    cert = "(snip)"
  }
}
```

I can work around this behavior by running this configuration separate from all my other ones, and specifying an empty `rootpath`.  But I expected `files` to work the same as in `vfat`.  Is this intended behavior or a bug?

thirtythreeforty, 2019-10-11:
If it's a bug, I'm happy to work on a fix.  Just wanted to know if I'm holding it wrong.

michaelolbrich, 2019-10-14:
I think you are correct. But I've not used this in a while. I'll talk to some more people to see what they think.

michaelolbrich, 2019-10-14:
I got some feedback, and we all agreed, that you are correct. So a patch would be great. Either copy the files to `tmpdir()/<something>` or use  `mountpath()` directly if no extra files are specified. Make sure the path is unique (maybe `rauc-<image-name>` or something like that, to ensure that creating two rauc bundles at the same time works as well.

PR #87 (2020-01-31): rauc fixes and test
Some rauc improvements.

Fixes: #77 
Fixes: #85
closed 2020-01-31 by michaelolbrich