pengutronix/genimage #84: Add support for exfat

erichiggins, 2019-12-15:
I was a bit surprised to see that the exfat file system was not supported by genimage. Would it be possible to add it? Since genimage is used by buildroot (and others), which frequently target embedded systems running on flash drives, it seems like a natural fit.

Thanks for all your hard work on genimage!

michaelolbrich, 2020-01-08:
It's a good idea, but not high on my personal priority list. But I'd be happy to review a pull request.

jameshilliard, 2020-04-09:
> Since genimage is used by buildroot (and others), which frequently target embedded systems running on flash drives, it seems like a natural fit.

To be fair Linux only recently added mainline kernel support for exfat(although I don't think it's possible to use exfat for a rootfs). I'm curious what the use case here would be as I'm not aware of any Linux systems capable of using exfat for storing/running bootloaders(fat32 being the most common standard there which supports large enough partitions for this purpose).

drewbug, 2020-04-21:
This would be useful for enabling certain end-users to add large files to a USB drive

drewbug, 2020-06-13:
`virt-make-fs` worked for my use-case.