pengutronix/genimage #91: error: ‘AT_NO_AUTOMOUNT’ undeclared

mentaal, 2020-02-05:
Hi All, 
I am not sure if this is the right place to post this , apologies in advance if it's not. 

I am using buildroot to build a linux environment for an FPGA platform.
It gets to the point where it attempts to build genimage which is where it's failing at:

```
 

        genimage 11
        =====

        prefix:                 /nobackup/user_g/tmp_build/host

        compiler:               /cad/apps/gnu/linux/x86_64/6/local/gcc/9.1.0/bin/gcc
        cflags:                 -O2 -I/nobackup/user_g/tmp_build/host/include  -pipe -Wall -Wextra -Wmissing-declarations -Wmissing-prototypes -Wnested-externs -Wpointer-arith -Wsign-compare -Wchar-subscripts -Wstrict-prototypes -Wshadow -Wformat-security -Wtype-limits -Wno-unused-parameter -ffunction-sections -fdata-sections -fvisibility=hidden
        ldflags:                -L/nobackup/user_g/tmp_build/host/lib -Wl,-rpath,/nobackup/user_g/tmp_build/host/lib  -Wl,--as-needed -Wl,--gc-sections

        debug:                  no
        hide symbols:           yes
        libconfuse:             -L/nobackup/user_g/tmp_build/host/lib -lconfuse

>>> host-genimage 11 Building
PATH="/nobackup/user_g/tmp_build/host/bin:/nobackup/user_g/tmp_build/host/sbin:/cad/apps/gnu/linux/x86_64/6/local/gcc/9.1.0/bin:/home/rs/lintel:/cad/local/bin:/cad/div/bin:/usr/cadtools/local/bin:/usr/cadtools/bin:/cad/bin:/cad/apps/gnu/linux/x86_64/6/bin:/usr/local/bin:/usr/local/sbin:/usr/sbin:/sbin:/usr/bin:/bin" PKG_CONFIG="/nobackup/user_g/tmp_build/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/nobackup/user_g/tmp_build/host/lib/pkgconfig:/nobackup/user_g/tmp_build/host/share/pkgconfig"  /usr/bin/make   -C /nobackup/user_g/tmp_build/build/host-genimage-11/
/usr/bin/make --no-print-directory all-am
  CC       genimage-genimage.o
  CC       genimage-config.o
  CC       genimage-util.o
  CC       genimage-crc32.o
  CC       genimage-image-cpio.o
util.c: In function ‘dir_size’:
util.c:648:36: error: ‘AT_NO_AUTOMOUNT’ undeclared (first use in this function); did you mean ‘S_AUTOMOUNT’?
  648 |   if (fstatat(fd,  d->d_name, &st, AT_NO_AUTOMOUNT) < 0) {
      |                                    ^~~~~~~~~~~~~~~
      |                                    S_AUTOMOUNT
util.c:648:36: note: each undeclared identifier is reported only once for each function it appears in
make[3]: *** [genimage-util.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [all] Error 2
make[1]: *** [/nobackup/user_g/tmp_build/build/host-genimage-11/.stamp_built] Error 2
make: *** [_all] Error 2

```

I am using `v11` apparently. 

Any suggestions welcome. 
I probably don't have enough environment information here so please let me know and I'll add it.

texierp, 2020-02-05:
Hi @mentaal,

> I probably don't have enough environment information here so please let me know and I'll add it.

Can you share the Buildroot version ?

Also it seems old kernel headers don't define this flag (what is your version ?).

Just for test purpose, can you try with this patch https://github.com/texierp/genimage/commit/8b040c90ffb7308020e1dec074b3a81f13af4491
(Just put the patch under buildroot/package/genimage)

Thanks !

mentaal, 2020-02-05:
Thanks very much for the prompt reply! Much appreciated. 
I'm still finding my way with buildroot so please bare with me. 

I tried the latest on master first and then I tried buildroot tag 2019.11.1.

I tried using the `zynqmp_zcu106_defconfig` default configuration and left everything as default. 

You're right in that I do appear to be using old kernel headers. Apparently 4.9.x is what I am using. I can see this in the aforementioned defconfig file:

`BR2_PACKAGE_HOST_LINUX_HEADERS_CUSTOM_4_9=y`

I'll try that patch shortly. Thanks!

mentaal, 2020-02-05:
@texierp I've been trying to apply that patch you mention but I'm having problems with it. This is just down to my understanding of buildroot I'm sure. 

Some more info about my environment first:

I've tried an "out of tree build" in that I am trying to build with `O=tmp_build`. Why did I do this? It just seemed like a handy thing to do as a starting point, to not pollute the buildroot parent directory too much and I will need to support multiple environments so it's something I have to get to grips with anyway. 

I tried placing the patch within `buildroot/package/genimage` as you suggested and then tried rebuild with make which doesn't work in that I don't see it applying the patch. 

After reading the buildroot training material I have (from bootlin) I see that my `BR2_GLOBAL_PATCH_DIR` points to `board/zynqmp/patches`. I tried creating a `board/zynqmp/patches/genimage/0001-util-define-AT_NO_AUTOMOUNT-if-needed.patch` which similarly didn't work. I also tried placing it at `board/zynqmp/patches/host-genimage` without success. 

I know this is off track from my initial question but I'm hoping this is something you just know off the top of your head...

mentaal, 2020-02-06:
Ah, nevermind, I have managed to apply the patch. I need to do a `dirclean` first. 

For what it's worth, the patch does indeed allow genimage to build.

closed 2020-02-06 by mentaal

michaelolbrich, 2020-02-10:
@texierp a pull request for the patch (with an extra comment about old kernel header) would be great.

texierp, 2020-02-10:
> @texierp a pull request for the patch (with an extra comment about old kernel header) would be great.

@michaelolbrich sure, I will do ;), I was waiting for confirmation from @mentaal 

Thanks !

PR #92 (2020-02-10): util: define AT_NO_AUTOMOUNT if needed
With old kernel headers, AT_NO_AUTOMOUNT flag is not defined.

Fixes #91
commit 42a161a8 (2020-02-10): util: define AT_NO_AUTOMOUNT if needed

With old kernel headers, AT_NO_AUTOMOUNT flag is not defined.

Fixes #91