pengutronix/genimage #98: config examples/docs

bcg62, 2020-03-27:
looking for an example configuration to install grub2 in a gpt image with a bios boot partition. So far none of my attempts have been bootable.

thanks

michaelolbrich, 2020-04-03:
I have no idea how that could work. As far as I know, grub does some magic during installation and need mounted partitions for this. But I think you need to ask the grub people if it's possible to install it into an image. If that's possible, then you could use a `exec-post` script to do it.

closed 2020-04-03 by michaelolbrich

Harvie, 2022-08-09:
Yes, this is rather annoying. I am also trying to create bootable x86 image using genimage and chroot made by debootstrap. It's not exactly straightforward task, especialy because `grub-mkconfig` can only create grub config for running OS (at least chroot with proper bind mounts). On the other hand `grub-install` seems to have some options to specify alternative paths, so in theory it might be possible to use `grub-install` with that chroot directory and then generate `grub.cfg` by custom scripts before generating the image. But you're kinda on your own with that. Would be better if `grub-mkconfig` allowed to specify boot/root partitions on commandline instead of autodetecting them from running system.