pengutronix/genimage #115: How to use compressed rootfs with this tool?

nmcain, 2020-08-12:
I am using the PC x86_64 Buildroot defconfig, with the following CFG configuration for genimage:
```shell
image efi-part.vfat {
  vfat {
    file startup.nsh {
      image = "efi-part/startup.nsh"
    }
    file EFI {
      image = "efi-part/EFI"
    }
    file bzImage {
      image = "bzImage"
    }
  }
  size = 16777216
}

image disk.img {
  hdimage {
    gpt = true
  }

  partition boot {
    image = "efi-part.vfat"
    partition-type-uuid = c12a7328-f81f-11d2-ba4b-00a0c93ec93b
    offset = 32768
    size = 16777216
    bootable = true
  }

  partition root {
    partition-type-uuid = 44479540-f297-41b2-9af7-d131d5f0458a
    partition-uuid = UUID_TMP
    image = "rootfs.ext2"
    offset = 16809984
  }
}
```
the file in question, rootfs.ext2 is set to a size of 500mb (I would love to decrease the size somehow)
However, I either want to compress the image or find some way of decreasing the size automatically. I attempted to use rootfs.ext2.xz, however it went into a kernel panic immediately.

michaelolbrich, 2020-08-12:
You can't just compress the rootfs. Linux can't boot from that. It your `rootfs.ext2` full? Maybe you can adjust the size in Buildroot. The Buildroot mailing list is probably a better place for this question. There is nothing genimage can do here.

closed 2020-08-12 by michaelolbrich

nmcain, 2020-08-12:
Genimage won't support a tar archive for Rootfs?

michaelolbrich, 2020-08-14:
I don't understand the question. You can put anything in a partition with genimage, including a tar archive. But that doesn't mean the Linux kernel can actually boot with it.