pengutronix/genimage #117: How to compile genimage on Ubuntu 18.04

Dizgah, 2020-08-13:
Hi there and excuse me for creating a new issue for my problem as this is a question, not an issue, but just because I did not find any mailing list here to ask my question I created this.
I'm intending to use genimage on Ubuntu 18.04 but I got `genimage: not found`.
I tried to install it on my OS but, there was not any candidate, even I tried to compile the last [release( V13)](https://github.com/pengutronix/genimage/releases/tag/v13)  But there is no build tutorial. I tried `./configure` to config the build and seems everything is OK but when I use `make` I got `make: *** No targets specified and no makefile found.  Stop.`

Bastian-Krause, 2020-08-13:
Hey,
try something like:
```
$ ./autogen.sh
$ ./configure
$ make
$ sudo make install
```

Dizgah, 2020-08-13:
@Bastian-Krause: Just `./configure` works for me, there is not any .`/autogen.sh ` script on sources of latest Release(V13) and also make failed!

Dizgah, 2020-08-14:
@Bastian-Krause:  I checkout current source( HEAD) which there is `autogen.sh` file there( I have no idea why it is missed on Release revesion) and when I run `autogen.sh` script it runs correctly! but when I run `./configure` I got:

```
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for gcc option to accept ISO C99... none needed
checking for gcc option to accept ISO Standard C... (cached) none needed
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... no
checking whether make supports nested variables... (cached) yes
checking for memset... yes
checking for setenv... yes
checking for strdup... yes
checking for strcasecmp... yes
checking for strerror... yes
checking for strstr... yes
checking for strtoull... yes
checking for inline... inline
checking for error_at_line... yes
checking for stdlib.h... (cached) yes
checking for GNU libc compatible malloc... yes
checking for size_t... yes
checking for uint32_t... yes
checking whether to enable debugging... no
checking whether to hide internal symbols... yes
checking if gcc supports -Wl,--as-needed flag... yes
checking if gcc supports -Wl,--gc-sections flag... yes
./configure: line 5689: syntax error near unexpected token `CONFUSE,'
./configure: line 5689: `PKG_CHECK_MODULES(CONFUSE, libconfuse)'
```
Do you have any idea what `CONFUSE` is ? and is it necessary or not? can I except it?
Because of the above error when I try to build using `make `command, I got:
`make: *** No targets specified and no makefile found.  Stop.`

michaelolbrich, 2020-08-14:
So, the release tarball does not contain the autogen.sh. This is a Bug, but it's not actually needed because the configure script is included here.

So if you use the tarball, then just start with `./configure`, and if you use git, the you need to run `./autogen.sh` first.

Looking at the error, I guess, that `/usr/share/aclocal/pkg.m4` is missing. You need to install `pkg-config` and probably `libconfuse-dev`.

Dizgah, 2020-08-14:
@michaelolbrich : I have installed them both before. After rebooting and cloning a new repo it fixed and compiled successfully.
It could be great to add a compilation guide section to documentation.
Many thanks for your help ;)

closed 2020-08-14 by Dizgah

AndreiCherniaev, 2023-01-25:
> $ ./autogen.sh
> $ ./configure
> $ make
> $ sudo make install

If you want to have genimage_0.1-1_amd64.deb package to be installed use [checkinstall](http://checkinstall.izto.org/checkinstall.git/)
```
$ ./autogen.sh
$ ./configure
$ make
$ sudo checkinstall --pkgversion="0.1" --pkgaltsource=https://github.com/pengutronix/genimage --requires="libconfuse"
```

In this case you can remove genimage using
`sudo dpkg -r genimage`

P.S. Before you need install [libconfuse](https://github.com/myd7349/libconfuse.cmake/pull/1):
```
git clone https://github.com/libconfuse/libconfuse
cmake -S libconfuse/ -B build_libconfuse/ -DCMAKE_BUILD_TYPE=Release
cmake --build build_libconfuse/ --parallel 
cd build_libconfuse
sudo checkinstall --pkgname="libconfuse" --nodoc --pkgversion="1" --default
# In this case you can remove libconfuse using
# sudo dpkg -r libconfuse
```