pengutronix/genimage #118: ERROR: hdimage(disk.img): part root size (366951937) too small for rootfs.ext2 (367001600)

nmcain, 2020-08-15:
I found that this tool supports the "autoresize" parameter in the configuration, and it seems that that is perfect for what I want to achieve, however, I am unable to compile an image successfully while using it.
```
image efi-part.vfat {
  vfat {
    file startup.nsh {
      image = "efi-part/startup.nsh"
    }
    file EFI {
      image = "efi-part/EFI"
    }
    file bzImage {
      image = "bzImage"
    }
  }
  size = 16777216
}

image disk.img {
  hdimage {
    gpt = true
  }

  partition boot {
    image = "efi-part.vfat"
    partition-type-uuid = c12a7328-f81f-11d2-ba4b-00a0c93ec93b
    offset = 32768
    size = 16777216
    bootable = true
  }

  partition root {
    partition-type-uuid = 44479540-f297-41b2-9af7-d131d5f0458a
    partition-uuid = UUID_TMP
    image = "rootfs.ext2"
    offset = 16809984

    autoresize = true
  }
size = 383778817
}
```
Using this configuration, I recieve the following error:
 ```ERROR: hdimage(disk.img): part root size (366951937) too small for rootfs.ext2 (367001600)```
It seems that using the autoresize option is a few MB off, am I misiterpreting it? What does the autoresize option actually do? Is it responsible for expanding the last partition when flashed?

michaelolbrich, 2020-08-17:
The `autoresize` option means that the partition will use the remaining space of the whole image. It will not change the content of the partition. In your case, I think the `rootfs.ext2` is too large for the space available in the partition.

michaelolbrich, 2020-11-17:
Only the image. This is only an option for genimage. Partition tables don't have such a feature, so it's not possible to do that. You need some kind of tool to do this manually after the image is written to the pen drive.

closed 2021-01-08 by michaelolbrich