pengutronix/genimage #120: How to auto fill sd card without expanding the image

mttbx, 2020-09-21:
I want to make the image and burn it into my sd card.
Suppose the size of the image is 20M, and of sd card is 32G.
I want to know how to make the image automatically fill the rest of the sd card without generate a 32G image.

michaelolbrich, 2020-11-17:
What exactly do you mean with 'fill the rest of the sd card'? What should happen with it? genimage can create empty partitions. And with a classic mbr and at most 4 partitions, this means that nothing is written there, so the image remains small. But you still need to create the filesystem after writing the image to the sd card.

Also, the sd card is probably not exactly 32G so you need to ensure that you get the exact size correctly.

mttbx, 2020-11-19:
What I mean is to create a 20M image. When I burn it into my 32G sd card, the last partition will fill the remaining space. Is there a way?

aaribaud, 2020-12-18:
This is a job for the system inside the image to perform on first run, not for genimage to try at build time.

For instance, on  Raspberry Pi OS they do it by
* having the content of /boot/cmdline.txt passed as command line to the Linux kernel
* Placing an init= option on this command line to point to a specific shell script
* Having that script resize the rootfs, reset the init= option in cmdline to run the actual init process and reboot after 5 seconds
Other hardware targets may have different ways of doing this, but basically, it's a run-time issue.

closed 2020-12-18 by mttbx