pengutronix/genimage #122: genimage 13 unable to create a disk image with more than 3 partitions with type defined

xogium, 2020-10-13:
As the title suggests, it seems that genimage 13 is unable to create a valid disk image and errors out when there are more than 3 partitions with their type defined, with hdimage being gpt.

ERROR: hdimage(sdcard.img): hybrid MBR partitions (5) exceeds maximum of 3.

I have 5 partitions of type linux filesystem in my gpt partition table:

* ssbl
* 2x rootfs
* rauc slot storage
* overlayfs

For what its worth this seems to have appeared only in version 13, because with version 12 it worked fine...

Aren't there ways of making protective mbr instead of hybrid ?

Either way this sounds like either genimage is right and this is not something valid, but should be possible with protective instead of hybrid, or that genimage has now a bug when the type of the partition is defined.

xogium, 2020-10-13:
Here is my config, in case it matters:

```
image sdcard.img {
        name = "sdcard-2020-09-22-21-gbdfbbc8"
        hdimage {
                gpt = "true"
        }
        partition state {
                offset = 0x14400
                size = 0x30000
                image = "/dev/null"
                in-partition-table = "no"
        }
        partition barebox-environment {
                size = 64K
                image = "/dev/null"
        }
        partition ssbl {
                partition-type = 0x83
                image = "barebox.bin"
                size = 1M
        }
        partition PrimaryRootfs {
                partition-type = 0x83
                image = "rootfs.ext4"
        }
        partition SecondaryRootfs {
                partition-type = 0x83
                image = "rootfs.ext4"
        }
        partition Rauc {
                partition-type = 0x83
                image = "rauc.ext4"
        }
        partition overlayfs {
                partition-type=0x83
                image = "overlayfs.ext4"
        }
```

Hope this helps.

a3f, 2020-11-16:
> For what its worth this seems to have appeared only in version 13, because with version 12 it worked fine...

v12 didn't have 013b22f26b73 `("Add support for Hybrid MBR partitions")`. Prior to that the `partition-type` was ignored for GPT images, so that's why it worked. Now, its existence means you want to hybrid MBR.

If you drop it, you should get `partition-type-uuid = "L"` and protective MBR, which are the defaults.
Is this not the case for you?

By the way, assuming this is about your STM32MP1 project, the TF-A doesn't like hybrid MBR, see https://git.pengutronix.de/cgit/DistroKit/commit/?id=3906b6b5f7f8b37d0cd3f5e3e39952571f7fa259

xogium, 2020-11-17:
Hi,
yes, it's the stm32mp1 based project.
Ah, damn it...

Is there no way to enforce protective mbr aside from dropping that commit ?
I remember that back then it did create protective mbr.

michaelolbrich, 2020-11-17:
Take a look at the `README.rst`. You need to set the `partition-type-uuid` for `gpt` not the `partition-type`. The you'll get a protective mbr instead of the hybrid mode.

xogium, 2020-11-17:
Oh !
I'm blind... Indeed, okay. My bad. I will set this. Thanks to both of you, my project had more or less hit a stop sign till then.

michaelolbrich, 2020-11-17:
I think there is nothing to do for me. We can reopen this if this does not work

closed 2020-11-17 by michaelolbrich

a3f, 2020-11-17:
> Is there no way to enforce protective mbr aside from dropping that commit ?

With "Drop it", I meant to drop the `partition-type`, not the commit. :-)