pengutronix/genimage #142: fat filesystem not clean

LongHairedHacker, 2021-03-02:
I'm using genimage v14 to generate harddisk image for an embedded system.
While testing something else I notet that if I flash on my device and run `fsck.fat` on the fat partitions it reliably reports errors.

Here is a mininal `genimage.cfg` and test-script to reproduce the issue:

```
image uboot.img {
    size = 64M
    vfat {
        label = "uboot"
    }
    mountpoint = "uboot"
}

image broken.img {
    hdimage {
        align = 1M
        extended-partition = 2
    }

    partition uboot {
        image = "uboot.img"
        partition-type = 0x0c
    }

}
```

```bash
#!/bin/sh

set -x

mkdir -p root/uboot/test1
mkdir -p root/uboot/test2
echo foo > root/uboot/test2/bar

genimage

LO_DEV=$(sudo losetup --partscan --show --find images/broken.img)
sudo fsck.fat -n ${LO_DEV}p1
sudo losetup -d $LO_DEV
```

If I run it on my machine I get:
```
$ ./test.sh
+ mkdir -p root/uboot/test1
+ mkdir -p root/uboot/test2
+ echo foo
+ genimage
INFO: cmd: "mkdir -p "/home/schumb/misc/genimage_test/tmp"" (stderr):
INFO: cmd: "rm -rf "/home/schumb/misc/genimage_test/tmp"/*" (stderr):
INFO: cmd: "mkdir -p "/home/schumb/misc/genimage_test/tmp"" (stderr):
INFO: cmd: "cp -a "/home/schumb/misc/genimage_test/root" "/home/schumb/misc/genimage_test/tmp/root"" (stderr):
INFO: cmd: "mv "/home/schumb/misc/genimage_test/tmp/root/uboot" "/home/schumb/misc/genimage_test/tmp/mp-uboot"" (stderr):
INFO: cmd: "mkdir "/home/schumb/misc/genimage_test/tmp/root/uboot"" (stderr):
INFO: cmd: "chmod --reference="/home/schumb/misc/genimage_test/tmp/mp-uboot" "/home/schumb/misc/genimage_test/tmp/root/uboot"" (stderr):
INFO: cmd: "chown --reference="/home/schumb/misc/genimage_test/tmp/mp-uboot" "/home/schumb/misc/genimage_test/tmp/root/uboot"" (stderr):
INFO: cmd: "find '/home/schumb/misc/genimage_test/tmp/root' -depth -type d -printf '%P\0' | xargs -0 -I {} touch -r '/home/schumb/misc/genimage_test/root/{}' '/home/schumb/misc/genimage_test/tmp/root/{}'" (stderr):
INFO: cmd: "mkdir -p "/home/schumb/misc/genimage_test/images"" (stderr):
INFO: vfat(uboot.img): cmd: "dd if=/dev/zero of="/home/schumb/misc/genimage_test/images/uboot.img" seek=67108864 count=0 bs=1 2>/dev/null" (stderr):
INFO: vfat(uboot.img): cmd: "mkdosfs  -n 'uboot' '/home/schumb/misc/genimage_test/images/uboot.img'" (stderr):
mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
INFO: vfat(uboot.img): cmd: "MTOOLS_SKIP_CHECK=1 mcopy -bsp -i '/home/schumb/misc/genimage_test/images/uboot.img' '/home/schumb/misc/genimage_test/tmp/mp-uboot'/* ::" (stderr):
INFO: hdimage(broken.img): adding partition 'uboot' (in MBR) from 'uboot.img' ...
INFO: hdimage(broken.img): writing MBR
+ sudo losetup --partscan --show --find images/broken.img
+ LO_DEV=/dev/loop23
+ sudo fsck.fat -n /dev/loop23p1
fsck.fat 4.1 (2017-01-24)
/TEST1/.
  Start (0) does not point to parent (2)
/TEST2/.
  Start (0) does not point to parent (4)
Leaving filesystem unchanged.
/dev/loop23p1: 4 files, 3/32695 clusters
+ sudo losetup -d /dev/loop23
```

Note the fsck output:
```
fsck.fat 4.1 (2017-01-24)
/TEST1/.
  Start (0) does not point to parent (2)
/TEST2/.
  Start (0) does not point to parent (4)
Leaving filesystem unchanged.
```

It does not seem to be an actual problem in this instance.
If I mount the partition and look at the files they seem fine.

Am I doing something wrong here?
Or is it safe to ignore the errors reported by fsck?

**additional infos:**
* Ubuntu 18.04.5
* mcopy (GNU mtools) 4.0.18

michaelolbrich, 2021-03-04:
So I did some testing. This is not directly a `genimage` problem. It happens because genimage calls `mcopy` with `-b`. If I remove that option, then everything is fine.

The documentation for `-b` says:
> Batch mode. Optimized for huge recursive copies, but less secure if a crash happens during the copy.

So I assumed it is save to use. I guess we should drop it. I don't think `genimage` is actually used for "huge recursive copies".

LongHairedHacker, 2021-03-04:
Thanks for the quick feedback.

From my point of view: If dropping `-b` makes the copy process slower, but in the end the file system doesn't have errors,
that would be an acceptable trade-off.

I'd like to use genimage in our CI pipeline at some point, generating the images used for our production systems.
Understandably some people will get nervous if those images include partitions that fail a basic fsck.

For now I added a `exec-post` to each of the fat partition images, that runs `fsck.fat -p` on the image.
The file system ends up clean on none of the files are missing or corrupted.

PR #151 (2021-04-13): image-vfat: drop '-b' from mcopy
It enables the 'Batch  mode'. It is optimized for huge recursive copies. This is not
really needed here and fsck.vfat complains about the resulting filesystem.

So drop the option and expand the existing test. Fail if fsck.vfat modifies the
filesystem.

Fixes #142
commit fe74ea15 (2021-04-13): image-vfat: drop '-b' from mcopy

It enables the 'Batch  mode'. It is optimized for huge recursive copies. This is not
really needed here and fsck.vfat complains about the resulting filesystem.

So drop the option and expand the existing test. Fail if fsck.vfat modifies the
filesystem.

Fixes #142
closed 2021-04-13 by michaelolbrich