pengutronix/genimage #144: board/ci20/genimage.cfg:29: no sub-section title/index for 'config'

NirajTummala, 2021-03-22:
INFO: cmd: "mkdir -p "/home/nirajtummala/buildroot/output/build/genimage.tmp"" (stderr):
INFO: cmd: "rm -rf "/home/nirajtummala/buildroot/output/build/genimage.tmp"/*" (stderr):
INFO: cmd: "mkdir -p "/home/nirajtummala/buildroot/output/build/genimage.tmp"" (stderr):
INFO: cmd: "cp -a "/tmp/tmp.tU65pvVkfR" "/home/nirajtummala/buildroot/output/build/genimage.tmp/root"" (stderr):
INFO: cmd: "find '/home/nirajtummala/buildroot/output/build/genimage.tmp/root' -depth -type d -printf '%P\0' | xargs -0 -I {} touch -r '/tmp/tmp.tU65pvVkfR/{}' '/home/nirajtummala/buildroot/output/build/genimage.tmp/root/{}'" (stderr):
INFO: cmd: "mkdir -p "/home/nirajtummala/buildroot/output/images"" (stderr):
INFO: hdimage(sdcard.img): adding partition 'uboot-spl' from 'u-boot-spl.bin' ...
INFO: hdimage(sdcard.img): adding partition 'uboot' from 'u-boot.img' ...
INFO: hdimage(sdcard.img): adding partition 'uboot-env' from 'uboot-env.bin' ...
INFO: hdimage(sdcard.img): adding partition 'rootfs' (in MBR) from 'rootfs.ext4' ...
INFO: hdimage(sdcard.img): writing MBR

NirajTummala, 2021-03-22:
It happens for every board not just ci20 ...I have tried all genimage versions in buildroot and the error message above is with genimage version 14

michaelolbrich, 2021-03-22:
This is a parser error while trying to read the config file. What's the content of the config file?

NirajTummala, 2021-03-22:
image sdcard.img {
    hdimage {
    }

    partition uboot-spl {
        in-partition-table = "no"
        image = "u-boot-spl.bin"
        offset = 512
    }

    partition uboot {
        in-partition-table = "no"
        image = "u-boot.img"
        offset = 14k
    }

    partition uboot-env {
        in-partition-table = "no"
        image = "uboot-env.bin"
        offset = 526k
    }

    partition rootfs {
        partition-type = 0x83
        image = "rootfs.ext4"
        offset = 2M
    }
}

michaelolbrich, 2021-03-22:
Right, so this is not fatal and I'm quite sure, that you can just ignore it. For `genimage` there is an optional `config {}` section. But in most cases that is not needed. From what I can tell, newer versions of `libconfuse` (the library used to parse the config file) print this warning when genimage tries to access this section.

I'll leave this issue open because we should handle this better. Probably check if the section is actually there, or something like that.

But you can just ignore the message. The image should be created correctly anyways.

NirajTummala, 2021-03-22:
Ok Thank you

PR #152 (2021-04-13): config: don't try to access non-existent section
If the config file does not include a 'config' section, the parsing
produces a warning: "no sub-section title/index for 'config'".

Silence this by checking if the section exists before accessing it.

Fixes #144
commit 086380a8 (2021-04-13): config: don't try to access non-existent section

If the config file does not include a 'config' section, the parsing
produces a warning: "no sub-section title/index for 'config'".

Silence this by checking if the section exists before accessing it.

Fixes #144
closed 2021-04-13 by michaelolbrich