pengutronix/genimage #154: ext4 e2fsck failed

mttbx, 2021-04-25:
I got this error:
`UNEXPECTED INCONSISTENCY; RUN fsck MANUALLY. (i.e., without -a or -p options) `
I use genimage to generate root file system. I used chroot into the file system and use apt to install some softwares. There's some softwares would cause this error.
Here's the cfg:
```
image boot.vfat {
  name = "boot"
	vfat {
		files = {
			"linux/zImage"
		}
	}
	size = 16M
}

image root.ext4 {
  name = "root"
	ext4 {
	  features = metadata_csum
	  #features = "extents,uninit_bg,dir_index,has_journal"
	}
	size = 600M
}

image sdcard.img {
	hdimage {
	}

	partition u-boot {
		in-partition-table = "no"
		image = "uboot/u-boot-sunxi-with-spl.bin"
		offset = 8192
		size = 2088960 # 512KB - 8192
	}

	partition boot {
		partition-type = 0xC
		bootable = "true"
		image = "boot.vfat"
	}
	
	partition rootfs {
		partition-type = 0x83
		image = "root.ext4"
		size = 0
	}
}
```

mttbx, 2021-04-25:
Here's some info, so what should I do?
```
E2fsprogs 1.06 (October 7, 1996)
================================

Fixed serious bug in e2fsck: if the block descriptors are bad, don't
smash the backup copies in ext2fs_close().  (The problem was that when
e2fsck -p discovered the problem, while it was closing the filesystem
and exiting, it was also blowing away the backup superblocks on the
disk, which was less than friendly.)  We now make it the case that we
only write out the backup superblock and the back block descriptors if
the filesystem is completely free from problems.

Fixed a bug in block_interate in the lib/ext2fs library which caused
e2fsck to fail on GNU Hurd-created filesystems.

Add support for Linux/FT's bootloader, which actually uses
EXT2_BOOT_LOADER, and sets its mode bits which caused e2fsck to want
to clear the inode.

Add support for the "A" (no atime update) attribute.  (Note: this
attribute is not yet in production kernels.)

The test suite is not automatically run when doing a "make all" from
the top level directory.  Users should manually run "make check" if
they wish to run the test suite.

Upon a preenhalt(), make the printed message more explicit that
running e2fsck "MANUALLY" means without the -p or -a options.

In e2fsck, if a disconnected inode is zero-length, offer to clear it
instead of offering to connect it to lost+found.

In e2fsck, if a filesystem was just unmounted uncleanly, and needs
e2fsck to be run over it, change e2fsck to explicitly display this
fact.

For dumpe2fs and e2fsck, cause the -V option to print out which
version of the ext2fs library is actually getting used.  (This will
help detect mismatches of using a 1.06 utility with a 1.05 library,
etc.)

Programmers' notes:
-------------------

EXT2_SWAP_BYTES was changed to EXT2_FLAG_SWAP_BYTES, which better fits
the naming convention.

In ext2fs_initialize(), make sure the description for the inode bitmap
is correctly initialize.

Fixed minor type typo in ext2fs_allocate_generic_bitmap();

```

michaelolbrich, 2021-04-25:
> ```
> image root.ext4 {
>   name = "root"
> 	ext4 {
> 	  features = metadata_csum
> 	  #features = "extents,uninit_bg,dir_index,has_journal"
> 	}
> 	size = 600M
> }
> ```

I'm pretty sure that this is not doing what you expect. Basically, you _replace_ the default features with `metadata_csum`. The result is probably a very strange filesystem.

You should use `use-mke2fs = true`. In that case `features = metadata_csum` is added to the default ext4 features.

Does your problem still occur with this?

mttbx, 2021-04-25:
Awesome! Thank you for your help!

closed 2021-04-25 by mttbx