pengutronix/genimage #157: make check: FAIL: test/basic-images.test 7 - mke2fs

Un1Gfn, 2021-05-01:
https://github.com/pengutronix/genimage/archive/v14.tar.gz

tar xf v14.tar.gz

cd genimage-14

autoreconf --version

    autoreconf (GNU Autoconf) 2.71
    ...

<details><summary>./autogen.sh</summary>

```
configure.ac:10: warning: The macro `AC_PROG_CC_STDC' is obsolete.
configure.ac:10: You should run autoupdate.
./lib/autoconf/c.m4:1666: AC_PROG_CC_STDC is expanded from...
configure.ac:10: the top level
configure.ac:78: warning: AC_LANG_CONFTEST: no AC_LANG_SOURCE call detected in body
./lib/autoconf/lang.m4:199: AC_LANG_CONFTEST is expanded from...
./lib/autoconf/general.m4:2894: _AC_LINK_IFELSE is expanded from...
./lib/autoconf/general.m4:2911: AC_LINK_IFELSE is expanded from...
lib/m4sugar/m4sh.m4:692: _AS_IF_ELSE is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
./lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
./lib/autoconf/general.m4:2270: AC_CACHE_CHECK is expanded from...
m4/attributes.m4:87: CC_CHECK_LDFLAGS is expanded from...
configure.ac:78: the top level
configure.ac:79: warning: AC_LANG_CONFTEST: no AC_LANG_SOURCE call detected in body
./lib/autoconf/lang.m4:199: AC_LANG_CONFTEST is expanded from...
./lib/autoconf/general.m4:2894: _AC_LINK_IFELSE is expanded from...
./lib/autoconf/general.m4:2911: AC_LINK_IFELSE is expanded from...
lib/m4sugar/m4sh.m4:692: _AS_IF_ELSE is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
./lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
./lib/autoconf/general.m4:2270: AC_CACHE_CHECK is expanded from...
m4/attributes.m4:87: CC_CHECK_LDFLAGS is expanded from...
configure.ac:79: the top level
configure.ac:10: installing 'build-aux/compile'
configure.ac:29: installing 'build-aux/config.guess'
configure.ac:29: installing 'build-aux/config.sub'
configure.ac:9: installing 'build-aux/install-sh'
configure.ac:9: installing 'build-aux/missing'
configure.ac:111: installing 'build-aux/tap-driver.sh'
Makefile.am: installing 'build-aux/depcomp'

----------------------------------------------------------------
Initialized build system. For a common configuration please run:
----------------------------------------------------------------

./configure CFLAGS='-g -O0' --prefix=/usr
```

</details>

<details><summary>./configure CFLAGS='-g -O0' --prefix=/usr</summary>

```
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
[... 912 lines cut ...]
7+0 records in
7+0 records out
3584 bytes (3.6 kB, 3.5 KiB) copied, 4.9522e-05 s, 72.4 MB/s
7+0 records in
7+0 records out
3584 bytes (3.6 kB, 3.5 KiB) copied, 4.4713e-05 s, 80.2 MB/s
exec-fail.config:5: no sub-section title/index for 'config'
INFO: cmd: "mkdir -p "/tmp/genimage-14/trash directory.basic-images.test/tmp"" (stderr+stdout):
INFO: cmd: "rm -rf "/tmp/genimage-14/trash directory.basic-images.test/tmp"/*" (stderr+stdout):
INFO: cmd: "mkdir -p "/tmp/genimage-14/trash directory.basic-images.test/tmp"" (stderr+stdout):
INFO: cmd: "cp -a "/tmp/genimage-14/trash directory.basic-images.test/root" "/tmp/genimage-14/trash directory.basic-images.test/tmp/root"" (stderr+stdout):
INFO: cmd: "find '/tmp/genimage-14/trash directory.basic-images.test/tmp/root' -depth -type d -printf '%P\0' | xargs -0 -I {} touch -r '/tmp/genimage-14/trash directory.basic-images.test/root/{}' '/tmp/genimage-14/trash directory.basic-images.test/tmp/root/{}'" (stderr+stdout):
INFO: cmd: "mkdir -p "/tmp/genimage-14/trash directory.basic-images.test/images"" (stderr+stdout):
INFO: file(file1.img): cmd: "false" (stderr+stdout):
ERROR: file(file1.img): failed to generate file1.img
  test_must_fail run_genimage exec-fail.config
ok 25 - exec-fail
PASS: test/basic-images.test 25 - exec-fail

# failed 1 among 25 test(s)
1..25
ERROR: test/basic-images.test - exited with status 1
```

</details>

Un1Gfn, 2021-05-01:
TL; DR - test failure w/ `mke2fs`?

Un1Gfn, 2021-05-01:
Current git master (g[7144925](https://github.com/pengutronix/genimage/tree/7144925279a435ba46655be73e0665f4832aaa41)) gives me `FAIL: test/basic-images.test 7 - mke2fs` as well.

michaelolbrich, 2021-05-01:
The new dump2fs produces different output for the same filesystem. This should be fixed with #155.

Un1Gfn, 2021-05-02:
So this is a bug with the test but not the app itself, and I can ignore it and make install?

michaelolbrich, 2021-05-02:
> So this is a bug in the test but not the app itself. It's ok to go ahead and `make install`. Is that correct?

Yes. I experimented a bit with old and new versions of e2fsprogs. As far as I can tell, the filesystem images are the same. It's just that tune2fs generates a different output for the same filesystem.

michaelolbrich, 2021-06-04:
Fixed by #155.

closed 2021-06-04 by michaelolbrich