pengutronix/genimage #158: Add BTRFS Support

liberodark, 2021-05-22:
Hi,

I want to build my project on BTRFS.
But genimage only support ext2, ext3, ext4...
Can you have possibility to add btrfs support.

Best Regards.

michaelolbrich, 2021-06-04:
I don't use btrfs myself so this has a low priority for me. But I would be happy to review a merge request.

liberodark, 2021-06-08:
Ok im poke @rtissera when you have time to check that.

lePereT, 2021-08-20:
This would be great to have. Is there any kind of guide other than the source code for how to add new filesystems + tests?

liberodark, 2022-03-06:
Any new about this feature ?

liberodark, 2022-05-17:
Hi, @Harvie 

I saw that you had added F2FS do you think you can add BTRFS ?

Best Regards

yodatak, 2022-11-12:
Hello some try to implent this i hope it help batocera ;) 

https://github.com/pengutronix/genimage/pull/206

Harvie, 2024-01-12:
> Hello some try to implent this i hope it help batocera ;)

Thanks for great work! It seems other guys have requested you to make few changes to your code in #206 before it can be merged to official genimage repository. This is not unusual thing to happen and it does help to maintain genimage codebase in top shape, so please don't get discouraged by it.

liberodark, 2024-01-18:
Thank you @yodatak 
Have make a PR fix try to merge this fonctionnality.
Based on https://github.com/pengutronix/genimage/pull/206

PR #264 (2024-09-26): Add Btrfs support
Based on #232, I've addressed the open comments there and expanded the test.

Closes #158.
Closes #232.
closed 2024-09-26 by michaelolbrich