pengutronix/genimage #162: Issue generating filesystem image

Graeme29, 2021-08-10:
Hello,

I'm compiling a Linux OS via 2021.05 Buildroot and using 5.10.57 version Linux Kernel.
When it's executing post-image script board/raspberrypi4-64/post-image.sh, i have an error : 
Disk full
INFO: vfat(boot.vfat): cmd: "rm -f "/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/boot.vfat"" (stderr):
ERROR: vfat(boot.vfat): failed to generate boot.vfat
make[1]: *** [Makefile:838 : target-post-image] Erreur 1
make: *** [Makefile:23 : _all] Erreur 2

Does anybody know how to solve the problem ?

Thanks for help, have a nice day !!!

Alexandre

> >>>   Executing post-image script board/raspberrypi4-64/post-image.sh
board/raspberrypi4-64/genimage-raspberrypi4-64.cfg:31: no sub-section title/index for 'config'
INFO: cmd: "mkdir -p "/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/build/genimage.tmp"" (stderr):
INFO: cmd: "rm -rf "/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/build/genimage.tmp"/*" (stderr):
INFO: cmd: "mkdir -p "/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/build/genimage.tmp"" (stderr):
INFO: cmd: "cp -a "/tmp/tmp.GJN41vc0Bx" "/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/build/genimage.tmp/root"" (stderr):
INFO: cmd: "find '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/build/genimage.tmp/root' -depth -type d -printf '%P\0' | xargs -0 -I {} touch -r '/tmp/tmp.GJN41vc0Bx/{}' '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/build/genimage.tmp/root/{}'" (stderr):
INFO: cmd: "mkdir -p "/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images"" (stderr):
INFO: vfat(boot.vfat): cmd: "dd if=/dev/zero of="/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/boot.vfat" seek=33554432 count=0 bs=1 2>/dev/null" (stderr):
INFO: vfat(boot.vfat): cmd: "mkdosfs   '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/boot.vfat'" (stderr):
INFO: vfat(boot.vfat): adding file 'bcm2711-rpi-4-b.dtb' as 'bcm2711-rpi-4-b.dtb' ...
INFO: vfat(boot.vfat): cmd: "MTOOLS_SKIP_CHECK=1 mcopy -bsp -i '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/boot.vfat' '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/bcm2711-rpi-4-b.dtb' '::'" (stderr):
INFO: vfat(boot.vfat): adding file 'rpi-firmware/cmdline.txt' as 'rpi-firmware/cmdline.txt' ...
INFO: vfat(boot.vfat): cmd: "MTOOLS_SKIP_CHECK=1 mcopy -bsp -i '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/boot.vfat' '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/rpi-firmware/cmdline.txt' '::'" (stderr):
INFO: vfat(boot.vfat): adding file 'rpi-firmware/config.txt' as 'rpi-firmware/config.txt' ...
INFO: vfat(boot.vfat): cmd: "MTOOLS_SKIP_CHECK=1 mcopy -bsp -i '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/boot.vfat' '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/rpi-firmware/config.txt' '::'" (stderr):
INFO: vfat(boot.vfat): adding file 'rpi-firmware/fixup.dat' as 'rpi-firmware/fixup.dat' ...
INFO: vfat(boot.vfat): cmd: "MTOOLS_SKIP_CHECK=1 mcopy -bsp -i '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/boot.vfat' '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/rpi-firmware/fixup.dat' '::'" (stderr):
INFO: vfat(boot.vfat): adding file 'rpi-firmware/start.elf' as 'rpi-firmware/start.elf' ...
INFO: vfat(boot.vfat): cmd: "MTOOLS_SKIP_CHECK=1 mcopy -bsp -i '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/boot.vfat' '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/rpi-firmware/start.elf' '::'" (stderr):
INFO: vfat(boot.vfat): adding file 'rpi-firmware/overlays' as 'rpi-firmware/overlays' ...
INFO: vfat(boot.vfat): cmd: "MTOOLS_SKIP_CHECK=1 mcopy -bsp -i '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/boot.vfat' '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/rpi-firmware/overlays' '::'" (stderr):
INFO: vfat(boot.vfat): adding file 'Image' as 'Image' ...
INFO: vfat(boot.vfat): cmd: "MTOOLS_SKIP_CHECK=1 mcopy -bsp -i '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/boot.vfat' '/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/Image' '::'" (stderr):
Disk full
INFO: vfat(boot.vfat): cmd: "rm -f "/home/heol/stage_heol/br-lab/64bit/systemctl/build-pi4-64/images/boot.vfat"" (stderr):
ERROR: vfat(boot.vfat): failed to generate boot.vfat
make[1]: *** [Makefile:838 : target-post-image] Erreur 1
make: *** [Makefile:23 : _all] Erreur 2

spockfish, 2021-08-18:
I think this is more of buildroot issue because of the 'Disk full' message then it has anything to do with genimage itself: when the disk is full of course genimage fails.

D3boker1, 2021-11-08:
> Hello,
>
> I'm compiling a Linux OS via 2021.05 Buildroot and using 5.10.57 version Linux Kernel. When it's executing post-image script board/raspberrypi4-64/post-image.sh, i have an error : Disk full [...]

Hi mate!
Do you solve your problem? 

If note, try to edit the geimage config file created by buildroot for your board.
In your case:

vim board/raspberrypi4-64/genimage-raspberrypi4-64.cfg

Change 32M for 64M, for example.

Best regards!