pengutronix/genimage #175: GPT:Primary headers thinks Alt. header is not at the end of the disk,

fabianbuettner, 2022-01-21:
Hi,
thanks for this great tool!
When booting the image, I created with genimage, I get the following warnings and I don't know how to get rid of them:

> GPT:Primary header thinks Alt. header is not at the end of the disk.
> GPT:8191999 != 15269887
> GPT:Alternate GPT header not at the end of the disk.
> GPT:8191999 != 15269887
> GPT: Use GNU Parted to correct GPT errors.

The genimage.cfg looks like this:

```
image boot.A {
        vfat {
                label = "A"
                files = {
                        "boot.scr",
                        "initramfs",
                        "devicetree",
                        "zImage"
                }
        }
        size = 16M
}

image boot.B {
        vfat {
                label = "B"
        }
        empty = true
        size = 16M
}

image emmc.img {
        size = 4000M
        hdimage {
                # Partition alignment. As per documentation defaults to 512 bytes
                partition-table-type = "gpt"
        }

        partition u-boot.env {
                # U-Boot environment GPT partition type
                partition-type-uuid = "3de21764-95bd-54bd-a5c3-4abe786f38a8"
		size = 32K
        }

        partition boot.A {
                # Basic data partition partition type
                partition-type-uuid = "ebd0a0a2-b9e5-4433-87c0-68b6b72699c7"
                bootable = true
                image = "boot.A"
                size = 16M
        }

        partition rootfs.A {
                # plain dm-crypt GPT partition type
                partition-type-uuid = "7ffec5c9-2d00-49b7-8941-3ea10a5586b7"
                image = "rootfs.img"
                size = 1500M
        }

        partition boot.B {
                # Basic data partition partition type
                partition-type-uuid = "ebd0a0a2-b9e5-4433-87c0-68b6b72699c7"
                bootable = true
                size = 16M
        }

        partition rootfs.B {
                # plain dm-crypt GPT partition type
                partition-type-uuid = "7ffec5c9-2d00-49b7-8941-3ea10a5586b7"
                size = 1500M
        }

        partition resources {
                # plain dm-crypt GPT partition type
                partition-type-uuid = "7ffec5c9-2d00-49b7-8941-3ea10a5586b7"
                image = "resources.img"
                autoresize = true
        }
}
```

Any feedback on this issue is appreciated!
best regards,
Fabian

michaelolbrich, 2022-01-28:
When using GPT then there is a header in sector 2 and more or less a copy of it (the 'Alt. header' from the message) in the last sector of the disk.

genimage does no know the exact size of your disk. So it used whatever size you give it and puts the 'Alt. header' at the very end.

Here you specified 4000M. In sectors that's 8192000 sectors. Counting usually starts at zero, so the last sector number is 8191999. So that's what you see in the message above.

Your device however is ~7G in size, so the kernel expects the 'Alt. header' at a different place. That's what the message is about.

So what you can do  is:

- Ignore the warning. The Alt. header is only a backup in case the primary header gets corrupted. You don't necessarily need it.
- use a tool, e.g. parted as suggested to fix the partition table when you boot the first time. You may also want to resize the last partition and it's content at that point
- set the correct size in the genimage config. Be careful with that though. Not all devices with the same official size have exactly the same actual size.

There is nothing more that genimage can do here. There is no real live autoresize feature for GPT.

closed 2022-01-28 by michaelolbrich

fabianbuettner, 2022-01-28:
Thank you for the explanation.

Setting the correct size helped me getting rid of this message.