pengutronix/genimage #176: Allow to duplicate and edit existing ex file systems images

forty, 2022-02-07:
It would be nice to duplicate existing file systems and just update some properties.

Maybe the request is a bit too generic, so here is what I have in mind concretely: I have a existing ext4 root fs. I want to include it twice in my image, so they can serve as active/inactive rootfs slot for RAUC. Ideally, they should have different UUID and LABEL. I don't find a way to do that with genimage only, even though that would be convenient.

To be honest, probably the only use case it to edit the label of existing ext (& vfat) file systems.

Example using an initial rootfs named rootfs.ext4

```
image rootfsa.ext4 {
  ext4 {
    label = "rootfsa"
    copy = "rootfs.ext4"
  }
  temporary = "true"
}

image rootfsb.ext4 {
  ext4 {
    label = "rootfsb"
    copy = "rootfs.ext4"
  }
  temporary = "true"
}

image sdcard.img {
  hdimage {
  }

  // bootloader partition skipped

  partition rootfsa {
    partition-type = 0x83
    image = "rootfsa.ext4"
  }

  partition rootfsb {
    partition-type = 0x83
    image = "rootfsb.ext4"
  }
}
```

michaelolbrich, 2022-02-08:
You can already do this with a `file` image and `exec-post`:
```
image rootfsa.ext4 {
  file {
    name = rootfs.ext4 
    copy = true
  }
  temporary = true
  exec-post = 'tune2fs -L "rootfsa" -U ac6aff85-6503-43ad-9bac-aea36dc86420 "${IMAGEOUTFILE}"'
}

image rootfsb.ext4 {
  file {
    name = rootfs.ext4 
    copy = true
  }
  temporary = true
  exec-post = 'tune2fs -L "rootfsb" -U 3523b1c0-eea8-4ae7-8ac7-f3203ae0d08a "${IMAGEOUTFILE}"'
}
```

For vfat you can use `fatlabel`. You could probably even change the content with `debugfs` and `mcopy`.

forty, 2022-02-08:
Ah great, I hadn't thought of doing it that way, I think that works well, thanks!

closed 2022-02-08 by forty