pengutronix/genimage #178: no such option 'partition-table-type' error

Sourabh1107, 2022-03-03:
Hello,

I am facing an issue while using `genimage.cfg` in one of my project with Buildroot build system. My `genimage.cfg` looks like:

```
image sdcard.img {
        hdimage {
                partition-table-type = "mbr"
                extended-partition = 3
        }

        partition ubootenv0 {
                image = "uboot-env.bin"
                in-partition-table = false
                offset = 1M
        }
```

But I am getting an error like:

```
[...]/genimage.cfg:41: no such option 'partition-table-type'
make[1]: *** [Makefile:827: target-post-image] Error 1
make: *** [Makefile:23: _all] Error 2
```

The `genimage.cfg` is part of `post-image.sh` and is called like `GENIMAGE_CFG="${BOARD_DIR}/genimage.cfg"`.

Can anyone please let me know how to resolve this issue?

Your help will be much appreciated.

Thanks in advance.

michaelolbrich, 2022-03-03:
Maybe your genimage version is too old? This option was introduced in version 15. Use `gpt = true` for older genimage versions.

Sourabh1107, 2022-03-03:
Hello @michaelolbrich Thanks for the quick reply. Yes, I think so.

The `genimage --version` returned 14. How can I upgrade version 15?

michaelolbrich, 2022-03-03:
That needs to happen somewhere in Buildroot. So you need to talk to the Buildroot people about that.

closed 2022-03-03 by michaelolbrich

Sourabh1107, 2022-03-03:
Hello @michaelolbrich I changed `partition-table-type = "mbr"` to `gpt = true` and also included `partition-type-uuid = "L"`. But again I am getting `no such option 'partition-type-uuid'`.

If I remove `partition-type-uuid = "L"`, then I am getting:

```
INFO: hdimage(sdcard.img): adding partition 'ubootenv0' from 'uboot-env.bin' ...
INFO: hdimage(sdcard.img): adding partition 'ubootenv1' from 'uboot-env.bin' ...
INFO: hdimage(sdcard.img): adding partition 'boot0' (in MBR) from 'boot.vfat' ...
INFO: hdimage(sdcard.img): adding partition 'boot1' from 'boot.vfat' ...
INFO: hdimage(sdcard.img): adding partition 'rescue' (in MBR) from 'rootfs.squashfs' ...
INFO: hdimage(sdcard.img): adding partition 'rootfs0' (in MBR) from 'rootfs.ext4' ...
INFO: hdimage(sdcard.img): writing EBR
INFO: hdimage(sdcard.img): adding partition 'rootfs1' (in MBR) from 'rootfs.ext4' ...
INFO: hdimage(sdcard.img): writing EBR
INFO: hdimage(sdcard.img): adding partition 'upload' (in MBR) from 'upload.ext4' ...
INFO: hdimage(sdcard.img): writing EBR
INFO: hdimage(sdcard.img): adding partition 'data' (in MBR) from 'data.ext4' ...
INFO: hdimage(sdcard.img): writing EBR
INFO: hdimage(sdcard.img): writing GPT
ERROR: hdimage(sdcard.img): hybrid MBR partitions (6) exceeds maximum of 3
INFO: hdimage(sdcard.img): cmd: "rm -f "/root/raspcm4/sources/output/images/sdcard.img"" (stderr):
ERROR: hdimage(sdcard.img): failed to generate sdcard.img
make[1]: *** [Makefile:827: target-post-image] Error 1
make: *** [Makefile:23: _all] Error 2
```

Genimage.cfg looks like:

```
image sdcard.img {
	hdimage {
		#partition-table-type = mbr
                gpt = true
		extended-partition = 3
	}

	partition ubootenv0 {
		image = "uboot-env.bin"
		in-partition-table = false
		offset = 1M
	}

	partition ubootenv1 {
		image = "uboot-env.bin"
		in-partition-table = false
		offset = 2M
	}

	partition boot0 {
		partition-type = 0xC
		bootable = true
		image = "boot.vfat"
		# Leave room for U-Boot environment
		offset = 4M
	}

	partition boot1 {
		image = "boot.vfat"
		in-partition-table = false
		# 256M + 4M
		offset = 260M
	}

	partition rescue {
		partition-type = 0x83
		image = "rootfs.squashfs"
		size = 256M
	}

	partition rootfs0 {
		partition-type = 0x83
		image = "rootfs.ext4"
		size = 900M
	}

	partition rootfs1 {
		partition-type = 0x83
		image = "rootfs.ext4"
		size = 900M
	}

	partition upload {
		partition-type = 0x83
		image = "upload.ext4"
		size = 900M
	}

	partition data {
		partition-type = 0x83
		image = "data.ext4"
		size = 128M
	}
}
```
Is there any other way to with older version of genimage?

Your help will be much appreciated.

michaelolbrich, 2022-03-03:
Where did you add the `partition-type-uuid`? It should be supported in version 14. And don't use `partition-type` with GPT. It enables the hybrid method and that does not allow more than 3 partitions.

Sourabh1107, 2022-03-03:
Hello @michaelolbrich I have set `partition-type-uuid = "L"` in

```
hdimage {
		#partition-table-type = mbr
                gpt = true
                partition-type-uuid = "L"
		extended-partition = 3
	}
```

Is it safe to remove `partition-type = 0x83` ? What effect will it have?

michaelolbrich, 2022-03-03:
The `partition-type-uuid` is the GPT equivalent  to `partition-type` for MBR. So replace `partition-type` with `partition-type-uuid` in each partition and don't set it in the `hdimage` section.