pengutronix/genimage #179: marry genimage and swupdate

flatmax, 2022-03-08:
[swupdate](https://github.com/sbabic/swupdate) is a tool for building firmware update images for embedded devices.

It would be nice to marry genimage with swupdate in some way. Perhaps the first step is to be able to output the genimage.cfg file as a genimage.json file (with JSON formatting).

Is there are function in the code to dump the parsed genimage.cfg information as text ? If so then perhaps that can be adapted to generate a valide json version of genimage.cfg files for use with swupdate.

michaelolbrich, 2022-03-08:
I don't know swupdate, but I expect this needs to be a bit more intelligent than just converting the config directly into json.
Probably a new image-swupdate.c that generates the config for swupdate and then calls it. We do something like that in image-rauc.c (That's also a firmware update tool).

Harvie, 2022-03-25:
Genimage can already create swupdate bundles. Swupdate uses CPIO format, which can be created by genimage.