pengutronix/genimage #181: How to run without a --rootpath

rossburton, 2022-03-17:
I'm building an image which doesn't have any root filesystems to populate: it's all firmware binaries and a kernel with initramfs.  However, I still need to either have a `root` directory or set `--rootpath` to an empty directory.

Is there a way in the config file to tell genimage that it doesn't need to copy any files?

Harvie, 2022-03-18:
Yes! This is so much needed.
I have prebuilt everything (debootstrap, blobs, bootloaders, etc...) in separate directories and i find it really annoying that genimage forces me to copy that to some --tmp directory. Can't you just take the files from where they already are?

This rootpat/mountpoint concept seems unnecessary to me.

I just want to eg. build ext4 filesystem from files which are already prepared in directory which i want to specify as relative path to genimage.cfg or current working directory. That is all i need, but it is not possible.

genimage instead forces me to copy all files from rootpath to tmp even when some of these files don't make sense to be copied at all.

Harvie, 2022-03-18:
I kinda found an ugly workaround:

Config like this:

```javascript
image image.cpio {
  cpio {
    format = "crc"
    extraargs = "-L"
  }
  mountpoint = "/../../image_contents"
}
```
Run like this:

```bash
mkdir fake_empty_root
./genimage --rootpath "./fake_empty_root" --inputpath "." --outputpath "."
```

This will
  * Create empty directory `fake_empty_root` and consider it root
  * Copy that empty directory to `tmp` directory
  * Run `cpio` to include `tmp/root/../../image_contents`
  * That way nothing is copied to `tmp`, but files in `./image_contents/` get in `image.cpio`

This is ugly but seems to work

Harvie, 2022-03-18:
Turned out to be very simple to implement. I've made PR, so we can discuss real code.
Probably might want to update README to reflect the change and come up with something better than `srcdir`, exact name of parameter can be changed easily.

michaelolbrich, 2022-03-21:
Hmm, I didn't test it but I think the problem with rootpath is in `collect_mountpoints()`. It should be possible to change this and only require the rootpath when at least one filesystem needs it.

Maybe set `image->empty` to true for stuff like hdimage and anything else that doesn't need the rootpath. And check for that and whatever we do in #184 to check if rootpath is actually needed.

closed 2022-03-25 by michaelolbrich

michaelolbrich, 2022-03-25:
With the current master, `--rootpath` is now optional if:

- none of the images actually need it, such as hd images etc.
- an image that would use it has an explicit `srcpath` specified.

rossburton, 2022-03-25:
That commit doesn't solve the issue I raised though.  If I have a `pack.config` which is about as minimal as possible:

```
image pack.img {
    hdimage {
        partition-table-type = "none"
    }
    partition 1 {
        image = "file1"
    }
}
```
Where `file1` is a file on disk, I still need to create or pass `--rootpath`:
```
$ genimage --config pack.config --inputpath .
INFO: cmd: "mkdir -p "/home/ross/Code/drive-by/genimage/tmp"" (stderr):
INFO: cmd: "rm -rf "/home/ross/Code/drive-by/genimage/tmp"/*" (stderr):
INFO: cmd: "mkdir -p "/home/ross/Code/drive-by/genimage/tmp"" (stderr):
INFO: cmd: "cp -a "/home/ross/Code/drive-by/genimage/root" "/home/ross/Code/drive-by/genimage/tmp/root"" (stderr):
cp: cannot stat '/home/ross/Code/drive-by/genimage/root': No such file or directory
```

rossburton, 2022-03-25:
@michaelolbrich your comment appeared whilst I was adding my comment.  Why does my example still need rootpath?

michaelolbrich, 2022-03-25:
Are you sure that you have the latest master? I just tested your config and it works as expected.

PR #184 (2022-03-25): Override filesystem source path
Fixes #181

Allows user to override the complexity of rootpath/mountpoint and just simply use existing directory at arbitrary path if needed.
Now i can create ext4 filesystem like this:

```
image root.img {
  ext4 {
    label = "rootlabel"
  }
  size = 1024M
  srcdir = "./root"
}
```

`genimage --inputpath "." --outputpath "."`