pengutronix/genimage #182: Add f2fs support

Harvie, 2022-03-18:
Hello,
i would like to use f2fs partitions in my images. would that be possible to add please?

Harvie, 2022-03-21:
I've been doing some research and f2fs-tools come with tool called `sload.f2fs` which allows to load files to filesystem image without mounting it. so i guess implementing this would be quite similar to jffs2 for example. but we need to call both `mkfs.f2fs` and `sload.f2fs`.

michaelolbrich, 2022-03-21:
That sound like it's technically possible. But I probably won't have time for it any time soon. But I would be happy to review a pull request for this.

Harvie, 2022-03-21:
Ok, i might take a look at it today. In the meantime, would you be so kind to take a look at my other PR, which is already pending? :-)

Harvie, 2022-03-21:
Yes, now i can confirm that `images/test.f2fs` does not exist by the time mkfs is called. weird.

Harvie, 2022-03-21:
Now i see that mkfs.ext2 does create the image file if it does not exist, while mkfs.f2fs does not create it.
@michaelolbrich should i just call fallocate in such case?

Harvie, 2022-03-21:
Ok, so i've copied code from image-vfat which solves exactly the same thing.

Harvie, 2022-03-21:
@michaelolbrich Ok, this PR seems to be working fine: https://github.com/pengutronix/genimage/pull/185

closed 2022-03-25 by michaelolbrich