pengutronix/genimage #191: Create partition with MDRAID (mdadm) array metadata

Harvie, 2022-05-02:
I have buildscript which creates harddisk image with partitions and some of them have MDRAID metadata on them.
I would like to migrate it to genimage, but there is no such feature.

_Side note: All i need is very simple setup. RAID1 with single(!) drive in it. It seems stupid to have RAID1 with only single drive. But it is much easier to grow such array later to desired state, when compared to converting plain non-RAID filesystem partition to RAID. I am usualy using this on routers which get deployed from the image in bulk. user can later easily add second SSD to the most important machines and grow the RAID across it once the deployment becomes mission-critical._

I've already tried to figure out how mdadm creates the metadata, but it seems it leaves metadata creation to kernel, which is bit annoying, because there is probably no userspace utility which can be called to do that offline without starting the array.

But there is `struct mdp_superblock_s` in header file `#include <md_p.h>`, so that can be included and populated from userspace. There seems to be mild complexity to it, but it should be possible under 200 lines of C code.

https://docs.huihoo.com/doxygen/linux/kernel/3.7/md__p_8h_source.html
https://docs.huihoo.com/doxygen/linux/kernel/3.7/structmdp__superblock__s.html

michaelolbrich, 2022-05-06:
Hmm, this is not exactly trivial and I have no idea about raid

Harvie, 2024-01-12:
OK, so i've made bit of progress.
I've put together following code that is able to generate "valid" mdraid RAID1 superblock, that is recognized by mdadm.

https://github.com/Harvie/Programs/blob/master/c/mdraid-gen/mdraid.c

It seems that now we just need to clean up few small things, set the data offsets correctly and append data to generated superblock with these offsets.

Harvie, 2024-12-19:
I've commited preliminary version of genimage with MD raid support to my fork, needs some testing tho... I would be happy to get some feedback on it...

Harvie, 2024-12-19:
This is example config that creates GPT image containing MDRAID partition containing EXT4 filesystem

```ini
image mdraid-hd.img {

  hdimage {
    partition-table-type = "gpt"
  }

  partition bootfs1 {
    image = "mdraid-md.img"
    partition-type-uuid = R
    bootable = "true"
    offset = 8M
    size = 6M
  }

}

image mdraid-ext4.img {
  ext4 {
    label = "ROOT_FS"
    use-mke2fs = true
  }
  size = 5M
}

image mdraid-md.img {
  mdraid {
    level = 1
  }
  partition data {
    image = "mdraid-ext4.img"
  }
}
```

closed 2025-01-17 by michaelolbrich