pengutronix/genimage #200: genimage-16 test fails for "test/ext.test 5 - mke2fs"

sudipm-mukherjee, 2022-10-10:
genimage test fails on `armel`, `armhf`, `i386` and `mipsel` with the error:
```
expecting success: 
	run_genimage_root mke2fs.config mke2fs.ext4 &&
	check_ext images/mke2fs.ext4 mke2fs 33554432 mke2fs

--- /<<PKGBUILDDIR>>/test/mke2fs.0.dump	2022-09-02 08:18:57.000000000 +0000
+++ dump	2022-10-07 20:48:11.891730974 +0000
@@ -11,7 +11,7 @@
 Inode count:              8192
 Block count:              32768
 Reserved block count:     1638
-Free blocks:              26568
+Free blocks:              26566
 Free inodes:              8141
 First block:              1
 Block size:               1024
@@ -29,7 +29,7 @@
 Maximum mount count:      -1
 Last checked:             Sat Jan  1 00:00:00 2000
 Check interval:           0 (<none>)
-Lifetime writes:          141 kB
+Lifetime writes:          143 kB
 Reserved blocks uid:      0 (user root)
 Reserved blocks gid:      0 (group root)
 First inode:              11
@@ -51,13 +51,13 @@
 Journal start:            0
 
 
-Group 0: (Blocks 1-8192) csum 0x74a0 [ITABLE_ZEROED]
+Group 0: (Blocks 1-8192) csum 0x3f30 [ITABLE_ZEROED]
   Primary superblock at 1, Group descriptors at 2-2
-  Block bitmap at 3 (+2), csum 0x16cec4db
+  Block bitmap at 3 (+2), csum 0x8d67cdef
   Inode bitmap at 7 (+6), csum 0xb1052088
   Inode table at 11-522 (+10)
-  6093 free blocks, 1997 free inodes, 18 directories, 1997 unused inodes
-  Free blocks: 2100-8192
+  6091 free blocks, 1997 free inodes, 18 directories, 1997 unused inodes
+  Free blocks: 2102-8192
   Free inodes: 52-2048
 Group 1: (Blocks 8193-16384) csum 0x8fde [INODE_UNINIT, BLOCK_UNINIT, ITABLE_ZEROED]
   Backup superblock at 8193, Group descriptors at 8194-8194
not ok 5 - mke2fs
FAIL: test/ext.test 5 - mke2fs
#	
#		run_genimage_root mke2fs.config mke2fs.ext4 &&
#		check_ext images/mke2fs.ext4 mke2fs 33554432 mke2fs
#	

# failed 1 among 5 test(s)
1..5
ERROR: test/ext.test - exited with status 1
```
All the 4 architecure gives the same error. I have tested manually on a mipsel machine and can see the error.
The build logs are also linked from `https://tracker.debian.org/pkg/genimage`

michaelolbrich, 2022-10-17:
Huh, this looks like mkfs.ext4 produces a different output on 32-bit architectures. The filesystem is probably just fine, but genimage tries to be reproducible, so I'll need to look into this.

michaelolbrich, 2022-12-08:
Can you try #207? This works for me on i386 but I cannot test the other architectures.

sudipm-mukherjee, 2022-12-11:
On Thu, 8 Dec 2022 at 15:57, Michael Olbrich ***@***.***> wrote:
>
> Can you try #207? This works for me on i386 but I cannot test the other architectures.

Thanks @michaelolbrich
Will try it tomorrow on all arch and confirm.

-- 
Regards
Sudip

sudipm-mukherjee, 2022-12-13:
Tested on mips machine and the issue is fixed after adding the patches.

Thanks @michaelolbrich. If you want me to test on all arch then I can upload the changes to Debian experimental and then that will build it on the Debian build servers for all the arch.

michaelolbrich, 2022-12-13:
I've merged it because it definitely fixes something. I'm not sure what the packages are based on, but I won't have the time to make a release this year, so you will need to carry the patch either way.