pengutronix/genimage #202: Doesn't build on FreeBSD

yurivict, 2022-10-16:
... due to use of linux-specific headers:
```
#include <linux/fs.h> 
#include <linux/fiemap.h>
```

Please consider porting to BSD.

michaelolbrich, 2022-10-17:
So this is used to detect holes. It makes copying images more efficient and is used to create Android sparse images.
If you can life without these the easy solution is to just assume there are no holes. We already do that if the runtime detection fails.

yurivict, 2022-10-17:
Is there a compile-time option to disable these?

michaelolbrich, 2022-11-11:
I think #204 should fix this. Can you try? I don't have any BSD to test this so I can only fake it a bit.

yurivict, 2022-11-25:
Thanks for fixing it.

Now it breaks like this:
```
util.c:475:7: warning: implicit declaration of function 'fallocate' is invalid in C99 [-Wimplicit-function-declaration]
                if (fallocate(fd, FALLOC_FL_PUNCH_HOLE | FALLOC_FL_KEEP_SIZE,
                    ^
util.c:475:21: error: use of undeclared identifier 'FALLOC_FL_PUNCH_HOLE'
                if (fallocate(fd, FALLOC_FL_PUNCH_HOLE | FALLOC_FL_KEEP_SIZE,
                                  ^
util.c:475:44: error: use of undeclared identifier 'FALLOC_FL_KEEP_SIZE'
                if (fallocate(fd, FALLOC_FL_PUNCH_HOLE | FALLOC_FL_KEEP_SIZE,
```
```fallocate``` is Linux-only. Can this be done with ```posix_fallocate``` ?

michaelolbrich, 2022-12-16:
So `posix_fallocate` does not have the same semantics. I've created #208 that just skips `fallocate` and uses the existing fallback path. Can you test this?

commit c0224caf (2022-11-25): make Linux specific headers optional

This should make it possible to build genimage on *BSD.
Add a build test to fake having no linux/*.h headers.

Fixes: #202
PR #208 (2023-01-06): fix building without fallocate()
It is not available on BSD so add a configure check for it. We already have fallback code in case fallocate() fails, so reuse that when it is missing.

Fixes: #202
commit 6bd884ab (2023-01-06): fix building without fallocate()

It is not available on BSD so add a configure check for it. We already have
fallback code in case fallocate() fails, so reuse that when it is missing.

Fixes: #202
closed 2023-01-06 by michaelolbrich