pengutronix/genimage #219: How to use tar to get an extra compressed file of the image?

maovidal, 2023-05-16:
Hi everyone:

I'm having trouble to understand how to use https://github.com/pengutronix/genimage#tar 

My ultimate interest is to get a `*.tar.gz` file from an image. Let's say I have this:

```
image boot.vfat {
	vfat {
		files = {
			"efi-part/EFI",
		}
	}
	size = 16352K # 16MB - 32KB
}
```

I have tried this on the same `genimage.cfg`:

```
image boot.vfat {
	vfat {
		files = {
			"efi-part/EFI",
		}
	}
	size = 16352K # 16MB - 32KB
	tar {}
}
```

But `genimage` complains about receiving 2 images.

Then I tried:

```
image boot.vfat {
	vfat {
		files = {
			"efi-part/EFI",
		}
	}
	size = 16352K # 16MB - 32KB
}

image boot.vfat.tar.gz {
	tar {}
}
```
This last one produces the `boot.vfat` but also an empty `boot.vfat.tar.gz`. What is the correct way to use `tar`?

Thank you!

a3f, 2023-05-17:
Do you want the contents of the image in the tar.gz or the image itself in compressed form?

maovidal, 2023-05-17:
Hi @a3f 

Mi intention is/was to get that `boot.vfat`image file compressed.
I was able to achieve it using the  `exec-post` option.

However, later I was thinking that maybe I understood incorrectly the `tar` option. I didn't find any clue about how to use it (inspecting the source with my rudimentary knowledge) as passing other options like `files` or `image`  are rejected by `genimage` for `tar`.

What could be an example of its use case and hos to use it? Currently this is all the content of the documentation for the `tar` option: 

```
Generates a tar image. The image will be compressed as defined by the filename suffix.
```

Thank you!

a3f, 2023-05-19:
`exec-post = "gzip -k $IMAGEOUTFILE"` is the correct way to achieve that. The tar image is for creating a tarball containing the files in the `--rootpath`.