pengutronix/genimage #221: genimage replaces chunks of `0x0` bytes in passed flash image with chunks of `0xff` bytes

hudson-ayers-cruise, 2023-08-21:
I am finding that genimage modifies an image (passed as a filepath) by replacing a large chunk of 0s in the middle of the passed file with `0xff` bytes in the generated image.

Is this expected behavior for flash images? Is there a way to disable it? This is a reduced version of the config I am using:

```
flash xspi-64M-256K {
      pebsize = 256K
      numpebs = 256
      minimum-io-unit-size = 256K
}

image nor-mfg.img {
  flash { }
  flashtype = "xspi-64M-256K"

  partition ivt {
    in-partition-table = "no"
    image = "program_image.s32"
    offset = 0
    size = 512K
  }

  partition u-boot-primary {
    in-partition-table = "no"
    image = "appcode.s32"
    offset = 512K
    size = 1536K
  }
}
```

`appcode.s32` contains a chunk of 0s starting at offset 0x450 and going until 0x78040. In the produced `nor-mfg.img`, there are 0's from 0x80450 to 0x81000, then all 0xff's from 0x81000 to 0xf8000.

This is causing me issues because the passed image is signed.

hudson-ayers-cruise, 2023-08-21:
After some additional debugging, I have discovered this only manifests when appcode.s32 is exactly 1536K bytes in size

hudson-ayers-cruise, 2023-08-21:
The problem seems to be that appcode.s32 is a sparse file. genimage treats the sparse, all-zero regions of the file as equivalent to holes, and fills them with 0xff, while the program I am using to sign the image (correctly) treats them as `0x0` bytes. Is this intended behavior?

michaelolbrich, 2023-08-31:
That's a bug. This was implemented for hd images where the holes in the sparse files are preserved when e.g. a partition image is copied. But part of that code is shared with the flash image and causes this bug.

I'm pretty busy right now, but this should not be hard to fix. In just ignore the holes in `insert_image()` if the fill byte is not zero:
```
if (byte == 0)
  map_file_extents(...)
else
  whole_file_exent(...)
```

hudson-ayers-cruise, 2023-08-31:
Thanks for the reply. I'll put together a PR

PR #222 (2023-09-05): Guarantee 0-holes in sparse files are filled with `0` bytes in flash images
Fixes #221 

Tested this fix using the example that inspired #221, verified that holes in sparse input files are now correctly populated with 0-bytes in the generated flash image, and that nothing else in the generated image changed.
closed 2023-09-05 by michaelolbrich