pengutronix/genimage #227: Image size in % does not work

Zlika, 2023-10-18:
Hello,
I'm using genimage 15 (through buildroot 2022.02.8) and specifying an image size in % does not seem to work.
My configuration file is:
```
image boot.vfat {
	vfat {
		files = {
			"zImage",
			"socfpga_cyclone5_de10_standard.dtb"
		}
		file extlinux/extlinux.conf {
			image = "extlinux.conf"
		}
	}

	size = 20M
}

image sdcard.img {
	hdimage {
	}

	partition boot {
		partition-type = 0xc
		bootable = "true"
		image = "boot.vfat"
	}

	partition uboot {
		partition-type = 0xa2
		image = "u-boot-with-spl.sfp"
	}
}
```

This config file works, but if I replace the "size = 20M" with a percent value (e.g. "size=120%" or even "size=200%") I always have the following error message:
`mkdosfs: Attempting to create a too small or too large filesystem.`

michaelolbrich, 2023-10-18:
Ah, I see the issue. For vfat images there are two ways to provide content:

1. The rootpath / mountpount stuff also used by other filesystems
2. The special file / files properties

The automatic size calculation only takes the rootpath into account.

michaelolbrich, 2023-10-18:
Certainly. But I'm not sure when I'll find the time to do that. The fastest way would be to implement it yourself and open a pull request. Basically you need to modify `vfat_setup()` and there
```
if (image->size_is_percent) {
    for all  extra files
        image->size += file size
}
```