pengutronix/genimage #236: Root path might not be needed for vfat with file list

detly, 2024-02-21:
If I have this config:

```text
image usb.img {
    hdimage {
        partition-table-type = gpt
    }
    partition root {
        image = "root.img"
    }
}

image root.img {
    size = "64M"
    vfat {
        files = {
            "readme.md"
        }
    }
}
```

...then I need to invoke `genimage` with a valid `--rootpath`. I'm not sure why though, because (*I think*) all the necessary files are specified, and could just be copied straight to `tmppath` as per other filesystems discussed in #181.

If I use `empty = true` (as a child key of `root.img`) then indeed I can skip supplying or creating `rootpath`. This suggests to me that `vfat` partitions populated from a list of files could _also_ qualify for not needing a `rootpath`.

- genimage: 16
- OS: Ubuntu 23.10

michaelolbrich, 2024-02-23:
Right, in that specific case, the rootpath should not be needed. Unfortunately it's a bit more complex to fix.
If the `--rootpath` is specified or if the image has a explicit mountpoint or srcpath and any of that is missing, then an error must still occur.
And there are multiple places that must be changed to take that into account.

detly, 2024-03-05:
I encountered this while using genimage with the meson build system, which [does not provide much control over output directories](https://github.com/mesonbuild/meson/issues/2320) in the build directory. (I don't want to start an argument over that policy, I'm just giving context.) An extra challenge is the need for the root path to be copied into the temporary path (or is it the other way around?) and therefore not contain it.

This can be worked around by using a wrapper script, or maybe by "properly" integrating genimage with meson via a plugin or official support. So it's not a total incompatibility. I don't mind either way if you want to close this because you don't want to fix it, or leave it open to not lose track of it.