pengutronix/genimage #237: Temporary images cause tmppath to remain

detly, 2024-02-21:
Given this config:

```text
image usb.img {
    hdimage {
        partition-table-type = gpt
    }
    partition root {
        image = "root.img"
    }
}

image root.img {
    size = "64M"
    temporary = true
    vfat {
        files = {
            "readme.md"
        }
    }
}
```

...if I run `genimage --config usb.cfg` then the following files are generated:

```text
images/usb.img
root/ # empty
tmp/root.img
tmp/root/ # empty
```

If I then try to run `genimage` a second time, I get this output:

```text
$ genimage --config usb.cfg
ERROR: tmppath '/home/jason/Code/[project paths]/tmp' exists and is not empty
```

I can understand the competing goals to (a) leave temporary artefacts around for debugging after running, but also (b) not delete stuff that might be important by default at startup. But it makes it annoying to integrate with other tools if there's this extra cleanup logic that's only needed sometimes. So if it is deliberate and not just an oversight, I'd suggest a flag to control this eg. `--cleanup` or its opposite `--leave-tmp`, or something like that.

- genimage: 16
- OS: Ubuntu 23.10

michaelolbrich, 2024-02-23:
I cannot reproduce this here. genimage should delete the content of .../tmp at the end.

What's the output of the first genimage run? It should end with something like this:
```
INFO: cmd: "rm -rf "/home/jason/Code/[project paths]/tmp"/*" (stderr):
```

detly, 2024-02-23:
Here's the full output from a more minimal project:

```text
genimage-test ⚬ genimage --config usb.cfg 
INFO: cmd: "mkdir -p "/home/jason/Code/genimage-test/tmp"" (stderr):
INFO: cmd: "rm -rf "/home/jason/Code/genimage-test/tmp"/*" (stderr):
INFO: cmd: "mkdir -p "/home/jason/Code/genimage-test/tmp"" (stderr):
INFO: cmd: "cp -a "/home/jason/Code/genimage-test/root" "/home/jason/Code/genimage-test/tmp/root"" (stderr):
INFO: cmd: "mkdir -p "/home/jason/Code/genimage-test/images"" (stderr):
INFO: vfat(root.img): cmd: "mkdosfs   '/home/jason/Code/genimage-test/tmp/root.img'" (stderr):
INFO: vfat(root.img): adding file 'readme.md' as 'readme.md' ...
INFO: vfat(root.img): cmd: "MTOOLS_SKIP_CHECK=1 mcopy -sp -i '/home/jason/Code/genimage-test/tmp/root.img' '/home/jason/Code/genimage-test/input/readme.md' '::'" (stderr):
INFO: hdimage(usb.img): adding partition 'root' (in MBR) from 'root.img' ...
INFO: hdimage(usb.img): adding partition '[MBR]' ...
INFO: hdimage(usb.img): adding partition '[GPT header]' ...
INFO: hdimage(usb.img): adding partition '[GPT array]' ...
INFO: hdimage(usb.img): adding partition '[GPT backup]' ...
INFO: hdimage(usb.img): writing GPT
INFO: hdimage(usb.img): writing protective MBR
INFO: hdimage(usb.img): writing MBR
genimage-test ⚬ ls -sR
.:
total 20
4 images  4 input  4 root  4 tmp  4 usb.cfg

./images:
total 188
188 usb.img

./input:
total 0
0 readme.md

./root:
total 0

./tmp:
total 152
  4 root  148 root.img

./tmp/root:
total 0
genimage-test ⚬ genimage --config usb.cfg
ERROR: tmppath '/home/jason/Code/genimage-test/tmp' exists and is not empty
```

michaelolbrich, 2024-02-23:
Ah, in my test, tmppath already existed and in that case, the content was removed as expected. Try #239, that should fix this.

detly, 2024-03-05:
Yep, that works!

commit 942c3dd3 (2024-03-06): genimage: improve tmppath handling

Right now cleanup is partially broken:
If the tmppath exists, then its content is removed at the end as
expected. However, if the tmppath is missing then it is created but
nothing is removed.

Change this to remove the directory including its content at the end
when the directory was created but stick to removing just the content if
the directory existed before genimage was called.

Fixes: #237
closed 2024-03-06 by michaelolbrich