pengutronix/genimage #254: mke2fs tests failing

sudipm-mukherjee, 2024-06-25:
The tests for mkefs are failing on the latest Debian and Ubuntu images.
The test-suite.log shows:
```
expecting success: 
	run_genimage_root mke2fs.config mke2fs.ext4 &&
	check_ext images/mke2fs.ext4 mke2fs 33554432 mke2fs

--- /build/genimage/genimage/test/mke2fs.0.dump	2024-06-25 18:24:43.702152164 +0000
+++ dump	2024-06-25 18:25:21.309394925 +0000
@@ -29,11 +29,11 @@
 Maximum mount count:      -1
 Last checked:             Sat Jan  1 00:00:00 2000
 Check interval:           0 (<none>)
-Lifetime writes:          107 kB
+Lifetime writes:          110 kB
 Reserved blocks uid:      0 (user root)
 Reserved blocks gid:      0 (group root)
 First inode:              11
-Inode size:	          256
+Inode size:               256
 Required extra isize:     32
 Desired extra isize:      32
 Journal inode:            8
@@ -57,9 +57,9 @@
   6105 free blocks, 1997 free inodes, 18 directories, 1997 unused inodes
   Free blocks: 2088-8192
   Free inodes: 52-2048
-Group 1: (Blocks 8193-16384) csum 0x8fde [INODE_UNINIT, BLOCK_UNINIT, ITABLE_ZEROED]
+Group 1: (Blocks 8193-16384) csum 0x1510 [INODE_UNINIT, ITABLE_ZEROED]
   Backup superblock at 8193, Group descriptors at 8194-8194
-  Block bitmap at 4 (bg #0 + 3), csum 0x00000000
+  Block bitmap at 4 (bg #0 + 3), csum 0xc1d1d464
   Inode bitmap at 8 (bg #0 + 7), csum 0x00000000
   Inode table at 523-1034 (bg #0 + 522)
   8190 free blocks, 2048 free inodes, 0 directories, 2048 unused inodes
not ok 5 - mke2fs
FAIL: test/ext.test 5 - mke2fs
#	
#		run_genimage_root mke2fs.config mke2fs.ext4 &&
#		check_ext images/mke2fs.ext4 mke2fs 33554432 mke2fs
#	

# failed 1 among 5 test(s)
1..5
ERROR: test/ext.test - exited with status 1
```

michaelolbrich, 2024-06-25:
Right. I'm pretty sure this is caused by this change:
https://git.kernel.org/pub/scm/fs/ext2/e2fsprogs.git/commit/?id=7150bea307a30f393d184d81a80d9a3ae2e78638

So the created filesystem is not wrong, the mke2fs version just produces a somewhat different filesystem.

I think I need a new way to handle different e2fsprogs versions.

michaelolbrich, 2024-06-27:
Can you try #256?

closed 2024-06-28 by michaelolbrich

sudipm-mukherjee, 2024-06-28:
> Can you try #256?

Thanks @michaelolbrich. That worked. :)

PR #256 (2024-06-28): tests: ext: use different output files to compare to instead of trying to patch the output
Fixes #254