pengutronix/genimage #280: Check file existence only after running exec-pre

Harvie, 2025-01-21:
Lets say i want to generate some custom/proprietary/niche kind of image that is not supported in genimage and the support does not even make sense. I've found following works quite well. Please note i am demonstrating creation of ext4 image, which is obviously supported by genimage, but lets pretend it's not.

```ini
image file.img {
  file {}
  size = 5M
  name = "MYLABEL"
  exec-pre = "truncate -s $IMAGESIZE $IMAGEOUTFILE; mkfs.ext4 -L $IMAGENAME $IMAGEOUTFILE"
}
```

There is only one problem. `exec-pre` is not executed unless `file.img` already exists, so i need to do `touch file.img` before running genimage. I think it would be nice to have way of postponing the `stat()` after `exec-pre` OR providing something like `exec-gen` that would be executed beforehand.

What do you think about such change?

michaelolbrich, 2025-01-21:
I think we should make it explicit instead and have a custom type:

``` ini
image file.img {
  custom {
      exec = "truncate -s $IMAGESIZE $IMAGEOUTFILE; mkfs.ext4 -L $IMAGENAME $IMAGEOUTFILE"
  }
  size = 5M
  name = "MYLABEL"
}
```

michaelolbrich, 2025-06-12:
Implemented by #296.