pengutronix/genimage #282: How does partition autoresize work exactly?

learmj, 2025-01-29:
Hi. Very happy genimage user here. I have a question about using the ```autoresize``` partition option and would like some clarity on its use and why it's not working in my case, particularly when there are implicit ```file``` rules involved. Consider the following as an example:
```
image <IMAGE_DIR>/<IMAGE_NAME>.<IMAGE_SUFFIX> {
   hdimage {
      partition-table-type = "mbr"
   }

   partition boot {
      in-partition-table = true
      partition-type = 0xC
      image = boot.vfat
      bootable = true
   }

   partition root {
      in-partition-table = true
      partition-type = 0x83
      image = root.ext4
   }
}

image boot.vfat {
   vfat {
      label = "BOOT"
   }
   size = 50M
   mountpoint = "/boot/firmware"
}

image root.ext4 {
   ext4 {
      label = "ROOT"
      use-mke2fs = true
   }
   size = 500M
   mountpoint = "/"
}
```
Here, the resultant image size is defined by the size of both partition images plus additional necessary minimal headroom (eg partition table, padding, etc). I get an image that is approx 551MB. All good.

If I wanted to create a version of this image where the ext4 partition was expanded to fill a larger size, how would I do it? The documentation states:

**image**
> size:
> Size of this image in bytes. 'k', 'M' or 'G' can be used as suffix to specify the size in multiple of 1024 etc. The suffix 's' specifies a multiple of the (traditional) sector size of 512. If the image if filled from a mountpoint then '%' as suffix indicates a percentage. '200%' means the resulting filesystem should be about 50% filled. Note that is is only a rough estimate based on the original size of the content.

**partition**
> autoresize:	
> Boolean specifying that the partition should be resized automatically. For UBI volumes this means that the autoresize flag is set. Only one volume can have this flag. For hd images this can be used for the last partition. If set the partition will fill the remaining space of the image.

In order to expand (what I think autoresize means) the ext4 partition, genimage needs to know the size of the image I want to create. So I add ```size = 2G``` inside the top level image section ...

```
image <IMAGE_DIR>/<IMAGE_NAME>.<IMAGE_SUFFIX> {
   hdimage {
      partition-table-type = "mbr"
   }

   size = 2G

   partition boot {
   ...
}
```
...and add ```autoresize = true``` inside the root partition descriptor (the last one):
```
   partition root {
      in-partition-table = true
      partition-type = 0x83
      image = root.ext4
      autoresize = true
   }
```
Unfortunately, despite these additions, genimage doesn't generate an image that is 2G in size and doesn't do any resize. 

Unsurprisingly, I can't remove the ```size``` from the root.ext4 image while keeping the ```size``` in the top level image section:
```
ERROR: ext4(root.ext4): no size given or must not be zero
```
...and I can't use autoresize without size being in the top level section (even though it doesn't seem to do anything):
```
ERROR: hdimage(/path/to/work/xyz/artefacts/my.img): the image size must be specified when using an 'autoresize' partition
```

So my question is, does autoresize work and if so how? I'm able to manually resize root.ext4 simply by doing:
```
$ resize2fs work/xyz/artefacts/root.ext4 2G
resize2fs 1.47.0 (5-Feb-2023)
Resizing the filesystem on work/xyz/artefacts/root.ext4 to 2097152 (1k) blocks.
The filesystem on work/xyz/artefacts/root.ext4 is now 2097152 (1k) blocks long.
$
$ ls -lh work/xyz/artefacts/root.ext4
-rw-r--r-- 1 user group 2.0G Jan 29 13:13 work/xyz-min/artefacts/root.ext4
```

I fully appreciate that some filesystems (eg UBI) can do this, but what I'd like to accomplish is being able to resize my ext4 filesystem to fill the remaining space in the image. The documentation gives the impression that genimage would:

- Know that I'd like to create an image that's 2G
- See that I have autoresize set on the last partition, meaning I'd like to resize this partition to fill the remainder of the image up the size specified
- Generate the last partition image then resize it to fill the remaining space in the image
- Assemble the image

Please could you tell me how this can be accomplished?

Incidentally, the supported image types in the top level readme don't seem to be up to date with what's supported. For example, btrfs is supported but isn't mentioned in the readme. Should it be?

Thanks,
--  Matt

michaelolbrich, 2025-01-30:
By default, genimage does not write areas without content. The partition table should contain partitions that cover the whole 2G, it's just the image that is smaller. Once you write the image to e.g. an SD Card, the partitions are as expected.

If you want to use the image directly, use the `fill` option for the image. With that, the image will have the full size.

Note that the size of the resized partition is _not_ propagated to the creation of the ext4 image. So you would still need to resize the filesystem at runtime to match the partition size.

learmj, 2025-01-30:
Thanks @michaelolbrich okay, so ```autoresize``` applies only to the (last) partition, and not the _contents_ of that partition.

> The partition table should contain partitions that cover the whole 2G

What I observe:

If I specify ```size = 2G``` at the top level as described above, without any other changes to the config, the partition table does not indicate the root partition is larger than when ```size=2G``` is not specified. If I then add ```autoresize``` into the rootfs partition, the partition reports the expected partition as having a larger size, and inline with a 2G image size.

> If you want to use the image directly, use the fill option for the image. With that, the image will have the full size.

Understood.

> So you would still need to resize the filesystem at runtime to match the partition size.

Rather than performing an on-device resize of the filesystem image for the partition, I suppose it could be possible to use ```fill``` in the config, plus the image's ```exec-post``` hook to invoke ```resize2fs``` to resize the image, although calculating this new size would probably be quite difficult to do accurately.  Does that sound reasonable and would you have any recommendations for how to calculate the size to resize to? Lots of variables, I guess.

Cheers,
--  Matt

closed 2025-03-03 by learmj