pengutronix/genimage #285: Please support in-place writing of nested images

helmutg, 2025-03-23:
When writing an ext4 or vfat image to a partition, the filesystem image is first created and then copied. Given larger images (e.g. including Gnome), this poses a non-trivial slowdown (especially when the /tmp location is not yet tmpfs). Many of the filesystem management utils now support offesets and sizes and therefore support in-place creation in a larger image in principle.

For instance `mkfs.ext4` has `-E offset=...`. For `mtools`, the offset can be added to an image filename after `@@`.

jluebbe, 2025-03-24:
Alternatively, using [`FICLONERANGE`](https://man7.org/linux/man-pages/man2/ioctl_ficlonerange.2.html) when supported would give faster "copies" almost everywhere in genimage. I'd expect that we have the necessary alignment (4kiB, I think) in almost every case.

jluebbe, 2025-03-24:
> Or `copy_file_range()`. But I don't think either is supported by ext4, so while it would be a nice fallback, we should still implement writing in place.

As far as I can see, the flag to enable reflink/clone was never implemented for `copy_file_range()`, so `FICLONERANGE` is the better choice here.

Writing in place would target the first disk image using that filesystem? And copy to others from there? It seems that would be complex to implement.

michaelolbrich, 2025-03-24:
I would implement in place only if the fs image is used exactly once.

helmutg, 2025-03-24:
The downside of using `copy_file_range` and such is that intermediate images tend to be stored in `/tmp` which tends to be `tmpfs` these days and the target image tends to be stored elsewhere, so you're crossing a filesystem boundary here and as a result those nice kernel features may not apply.

I concur with optimizing the common case (image being used exactly once). What is requested here is an optimization and not expected to pose a semantic difference.

I recommend labeling the issue wishlist in any case.