pengutronix/genimage #291: [FeatureR] Add runtime option to change GPT_ENTRIES

Apaczer, 2025-05-13:
normally could be done via `sgdisk --resize-table=N [DEVICE]`, but here we need to patch `image-hd.c`

michaelolbrich, 2025-05-27:
Very strange. What tool do you use for the modification? I'm asking because we use the `gpt-location` to move the table behind the bootloader with other SoCs and resize and add partitions as well and I've never heard of a corrupted bootloader.

And have you tried it with a smaller table? I don't know how the corruption happens in your case, but the only thing I can think of is, that the tool moves the table to the default location. However, if it does that, the I would expect that is does not respect the reduced size either and increases the size back to 128 entries.

Apaczer, 2025-05-27:
> What tool do you use for the modification?

Mostly I'm trying to expand last partition to make use of the remaining space and I've tried `parted` to modify endpoint then other tools to resize (depends on FSTYPE). The gparted from my PC machine shows first 1MB sector as unallocated space, so I believe libparted fails to recognize relocated main GPT correctly (other than default).
I admit I've tested now with `gdisk` instead to modify partitions and it doesn't have this problem, so it was my fault sticking to `parted`.

> And have you tried it with a smaller table?

You mean reduces entries? Yes, with smaller value (without adjusting main partition table via `gpt-location`) the `parted` doesn't corrupts bootloader, so it might be still worth adding this functionality to make image more compatible/safe.